RIOT-OS/RIOT #2702: periph: spi: simplify spi_transfer_byte

kaspar030, 2015-03-24:
When working with SPI, I found ```spi_transfer_byte``` is overly complicated.

1. it can return an error, but does so only if the wrong SPI device is specified (in most implementations).
As ```simplify spi_transfer_byte``` is called by ```spi_transfer_bytes```, that check is repeated many times unnecessarily.

2. the docs say it returns "the number of bytes transferred". Shouldn't this always be 2? stm32f* implementations always return 1. samd21 returns 1 or two depending of the ```*in``` parameter.

I propose to change the signature to
```char spi_transfer_byte(spi_t dev, char out)```,
to skip dev checks, and to make it just return the byte that was read.

gebart, 2015-03-24:
In master mode there will (almost) never be any (detectable) transmission errors, there is however the possibility of detecting errors in slave mode. An earlier discussion (on the mailing list maybe?) mentioned that we have no use cases for SPI slave mode right now, but since we define an SPI slave API we should still consider the possibility of someone actually using it.

I still think this is good for master mode.

somewhat related:
 - Is there any platform which implements SPI as a DMA transfer or interrupt based transfers (in master mode, slave mode usually requires some interrupt handling regardless)?

PeterKietzmann, 2015-03-24:
According to "the number of bytes transferred" I agree that it is a bit useless for `spi_transfer_byte()`. In my opinion this function should always return `1` (that is probably the reason why stm32f* implementations do so) but I could also live with a `2` :-) . The normal procedure is that after pulling the CS-line low the master shifts its data on the bus through a shift register (1 Byte) and __at the same time__  reads data from the bus through the shift register (1 Byte). Many slave devices return a `STATUS` byte when the CS-line goes low. This is the byte that is read by the master. So actually there are two bytes transferred but (i) at the same time and (ii) the second byte will never be a requested information as it is send out each time the CS goes low.
Returning the read byte seems reasonable and logic to me but it feels a bit wrong comparing the "style" of our peripheral interfaces.

kaspar030, 2015-03-24:
@gebart wouldn't connecting two MCUs via SPI a master/slave situation / use-case?

@haukepetersen Any opinion?

haukepetersen, 2015-03-24:
yes! 

> it can return an error, but does so only if the wrong SPI device is specified (in most implementations).
As simplify spi_transfer_byte is called by spi_transfer_bytes, that check is repeated many times unnecessarily.

That is true, but keep in mind that the `spi_transfer` functions are not implemented very good, they basically are put together as simple as possible to work for now. In the long term, the implementations of these should use DMA and hence their implementation would not simply call `transfer_byter` many times...

Checking the SPI device should be included (arguably we could put a DEVELHELP guard around them, like done in #2701 by @gebart -> for this I don't really have an opinion). So the return value is needed!

> the docs say it returns "the number of bytes transferred". Shouldn't this always be 2?

False. You transfer 1 byte in each direction, but you do this concurrently and do not count both, so  `spi_transfer_byte` always returns 1 (if everything went fine).

So I am strictly against changing the API for `spi_transfer_byte` not only because of the points above, but also on behalf of keeping the peripheral APIs consistent!

kaspar030, 2015-03-24:
> So I am strictly against changing the API for spi_transfer_byte not only because of the points above, but
> also on behalf of keeping the peripheral APIs consistent!
would you be ok adding a ```spi_rw_raw()``` with the proposed signature?

haukepetersen, 2015-03-24:
nope, same problem: You get error when giving an invalid device identifier and you introduce functions that are not in-line with the rest of the perepheral API

LudwigOrtmann, 2015-03-24:
@kaspar030 if your goal is to be super-efficient, just hard-code your critical stuff ;)

kaspar030, 2015-03-24:
@haukepetersen Not checking the device identifier is FINE for a one byte reading function. For reading a register. Dammit. Can we call it "_unchecked"?
Also, there are other functions in the periph API which just return a value. Why not here?

@LudwigOrtmann Would be nice to be portable AND super-efficient... ;)

PeterKietzmann, 2017-01-31:
This issue is outdated. If there are concerns about the newly introduces SPI rework, please post them in #6437

closed 2017-01-31 by PeterKietzmann