RIOT-OS/RIOT #2724: Add support for serial number passing to CMSIS boards, document it

LudwigOrtmann, 2015-03-26:
For further information compare https://github.com/RIOT-OS/RIOT/pull/2318

> Hmm, when this gets merged the information on how to use it (i.e. how to discover and set the serial number) should probably be added to the wiki, right? But since this mechanism should work the same way for all CMSIS-DAP devices (?), should it really be added to the Sam R21 page or is there a better place to put it?

(Source: https://github.com/RIOT-OS/RIOT/pull/2318#issuecomment-86467444)

OlegHahm, 2015-09-02:
@x3ro, care to update the documentation accordingly?

x3ro, 2015-09-03:
Will do!

OlegHahm, 2015-10-01:
@x3ro, ping!

LudwigKnuepfer, 2015-10-25:
@x3ro ping!

x3ro, 2015-10-25:
My bad, have been postponing this for ever. It's first item on my todos list for tuesday!

x3ro, 2015-10-27:
Here's my draft, will add to the wiki when #4166 is merged.

# Connecting multiple boards

If you have multiple boards connected to the same computer and want to flash, this is what you can do:

First, run `make list-ttys`. This will show you a list of currently connected boards, their serial numbers and the names of their TTYs:

    Atmel Corp. EDBG CMSIS-DAP serial: 'ATML2127031800002124', tty(s): ttyACM0
    Atmel Corp. EDBG CMSIS-DAP serial: 'ATML2127031800002145', tty(s): ttyACM1

knowing your serial number, you can flash, open a terminal or debug a specific board like this:

    BOARD=samr21-xpro SERIAL="<serial>" make flash
    BOARD=samr21-xpro SERIAL="<serial>" make term
    BOARD=samr21-xpro SERIAL="<serial>" make debug

OlegHahm, 2015-10-29:
#4166 is  merged. @LudwigKnuepfer, @haukepetersen, are you fine with @x3ro's proposal.

@x3ro: maybe you could use the exemplary serial number from the exemplary `make list-ttys` output for the second example, too?

LudwigKnuepfer, 2015-10-29:
I'm fine, thanks.

PeterKietzmann, 2015-10-30:
@haukepetersen agree?  I assume we can close this issue then..

OlegHahm, 2015-10-31:
@PeterKietzmann, well, we should wait until @x3ro comes up with an according PR.

x3ro, 2015-11-02:
@OlegHahm Ich haette es jetzt einfach ins Wiki zu dem Board eingetragen. Woran hattest du gedacht? :)

OlegHahm, 2015-11-03:
Whatever feels most natural for you - you were the one asking for documentation first. ;)

x3ro, 2015-11-04:
Done: https://github.com/RIOT-OS/RIOT/wiki/Board:-SAMR21-xpro#connecting-multiple-boards

closed 2015-11-04 by x3ro