RIOT-OS/RIOT #2726: Investigate strange variable derefence pattern

LudwigOrtmann, 2015-03-27:
There are several occurrences like this:
```c
 (&GPIO_3_DEV)->OUTTGL.reg = 1 << GPIO_3_PIN;
```
1.
Figure out if this is necessary for some strange reason (look at the assembler).

2.
Find all the places where this is done with `git grep -E '\(&[^)]*\)->'` and fix them.

kushalsingh007, 2015-03-27:
I have started working on this issue :)

PeterKietzmann, 2015-07-03:
@kushalsingh007 are you still working on this? The occurrences may have changed after #3095. Didn't look for it, just came to my mind...

kushalsingh007, 2015-07-03:
Currently I have stopped working on it due to some other stuff that I am working on  .. But once I get free I'll have a look at it again. Meanwhile, if someone else wants have a look  .:+1:

MohmadAyman, 2015-08-29:
hello guys, i'm new to RIOT,
i'll start working on this ... hope it wont be hard for a newbie like me.

PeterKietzmann, 2015-08-31:
Hi @MohmadAyman and welcome :-) ! Nice that you're planning to start working on this issue. If you have any problems, just ask on the developers mailing list or directly in here. If there is progress you may  want to open a PR according to our [developement procedures](https://github.com/RIOT-OS/RIOT/wiki/Development-procedures). Good luck!

MohmadAyman, 2015-09-01:
https://github.com/RIOT-OS/RIOT/pull/3753

MohmadAyman, 2015-09-01:
i've created a PR, hope you guys check it soon, cause i'm not really sure of what i did.
https://github.com/RIOT-OS/RIOT/pull/3753

miri64, 2016-10-17:
So… what are the results of this?

OlegHahm, 2017-01-15:
@LudwigKnuepfer ?

smlng, 2017-07-06:
I think this can be closed, latest output for `git grep -E '\(&[^)]*\)->'` in RIOT:

```
cpu/lpc11u34/periph/spi.c:        *((volatile uint8_t *)(&dev(bus)->DR)) = tmp;
cpu/lpc11u34/periph/spi.c:        tmp = *((volatile uint8_t *)(&dev(bus)->DR));
cpu/msp430_common/include/stdatomic.h:    __atomic_compare_exchange_n(&(object)->__val, expected,     \
cpu/msp430_common/include/stdatomic.h:    __atomic_compare_exchange_n(&(object)->__val, expected,     \
cpu/msp430_common/include/stdatomic.h:    __atomic_exchange_n(&(object)->__val, desired, order)
cpu/msp430_common/include/stdatomic.h:    __atomic_fetch_add(&(object)->__val, operand, order)
cpu/msp430_common/include/stdatomic.h:    __atomic_fetch_and(&(object)->__val, operand, order)
cpu/msp430_common/include/stdatomic.h:    __atomic_fetch_or(&(object)->__val, operand, order)
cpu/msp430_common/include/stdatomic.h:    __atomic_fetch_sub(&(object)->__val, operand, order)
cpu/msp430_common/include/stdatomic.h:    __atomic_fetch_xor(&(object)->__val, operand, order)
cpu/msp430_common/include/stdatomic.h:    __atomic_load_n(&(object)->__val, order)
cpu/msp430_common/include/stdatomic.h:    __atomic_store_n(&(object)->__val, desired, order)
cpu/msp430_common/include/stdatomic.h:    (_Bool)((*__ep = __sync_val_compare_and_swap(&(object)->__val,  \
cpu/msp430_common/include/stdatomic.h:    __sync_swap(&(object)->__val, desired)
cpu/msp430_common/include/stdatomic.h:    __sync_lock_test_and_set(&(__o)->__val, __d);           \
cpu/msp430_common/include/stdatomic.h:    __sync_fetch_and_add(&(object)->__val,      \
cpu/msp430_common/include/stdatomic.h:    __sync_fetch_and_and(&(object)->__val, operand)
cpu/msp430_common/include/stdatomic.h:    __sync_fetch_and_or(&(object)->__val, operand)
cpu/msp430_common/include/stdatomic.h:    __sync_fetch_and_sub(&(object)->__val,      \
cpu/msp430_common/include/stdatomic.h:    __sync_fetch_and_xor(&(object)->__val, operand)
cpu/msp430_common/include/stdatomic.h:    __sync_fetch_and_add(&(object)->__val, 0)
pkg/lwip/contrib/sock/lwip_sock.c:                    return ip_2_ip4(&netif->ip_addr)->addr == ip_2_ip4(addr)->addr;
Binary file pkg/oonf_api/patches/0012-isolate-oonf-list-implementation-of-riot-s-implement.patch matches
sys/net/gnrc/link_layer/slip/gnrc_slip.c:        msg_send_int(&msg, _SLIP_DEV(arg)->slip_pid);
sys/net/gnrc/link_layer/slip/gnrc_slip.c:                if (ringbuffer_add_one(&_SLIP_DEV(arg)->in_buf, _SLIP_END) < 0) {
sys/net/gnrc/link_layer/slip/gnrc_slip.c:                if (ringbuffer_add_one(&_SLIP_DEV(arg)->in_buf, _SLIP_ESC) < 0) {
sys/net/gnrc/link_layer/slip/gnrc_slip.c:        if (ringbuffer_add_one(&_SLIP_DEV(arg)->in_buf, data) < 0) {
```

kaspar030, 2017-10-25:
> I think this can be closed, latest output for git grep -E '\(&[^)]*\)->' in RIOT:

Confirmed. I'll close.

closed 2017-10-25 by kaspar030