RIOT-OS/RIOT #2732: applications/sniffer freezes on SAM R21 when setting transceiver channel

x3ro, 2015-03-28:
After some debugging I've narrowed it down to [this line](https://github.com/RIOT-OS/applications/blob/master/sniffer/main.c#L138) (the first one):

    uint16_t c = 10;
    tcmd.transceivers = TRANSCEIVER_DEFAULT;
    tcmd.data = &c;
    printf("Set transceiver to channel %u\n", c);
    msg_send(&mesg, transceiver_pid);

I've compared it to how the transceiver channel is set in the `shell` (which works) and noticed that the type is a `uint32_t` there. After changing the type in the sniffer application, everything works fine. Possibly related to #2727.

Wouldn't it make sense to refactor the functionality in e.g. shell/transceiver to a utility API that allows this kind of code to be called from any RIOT application? I'm assuming that things like setting the transceiver channel are quite common.

LudwigOrtmann, 2015-03-28:
The transceiver module is obsolete.

OlegHahm, 2015-04-02:
I don't think that this is related to the `transceiver` module. And it is still the only possibility to run 6LoWPAN on RIOT. I'm reluctant to call anything obsolete that has no alternative already in place.

OlegHahm, 2015-08-17:
Now it is obsolete.

closed 2015-08-17 by OlegHahm