RIOT-OS/RIOT #2748: Native platform not working

cladmi, 2015-03-31:
The native platform examples are raising a segmentation fault on my computer on `master`

    /home/harter/work/iot-lab/parts/RIOT/examples/hello-world/bin/native/hello-world.elf  
    make: *** [term] Segmentation fault

Bisecting the issue pointed to this commit: 00092e8308311a1221c382ddc6410c297f12b3de

The `valgrind` trace has the following errors:

```
==28538== Jump to the invalid address stated on the next line
==28538==    at 0x0: ???
==28538==    by 0x804B091: native_hwtimer_pre_init (in /home/harter/work/iot-lab/parts/RIOT/examples/hello-world/bin/native/hello-world.elf)
==28538==    by 0x80493E0: startup (in /home/harter/work/iot-lab/parts/RIOT/examples/hello-world/bin/native/hello-world.elf)
==28538==    by 0x804D731: __libc_csu_init (in /home/harter/work/iot-lab/parts/RIOT/examples/hello-world/bin/native/hello-world.elf)
==28538==    by 0x4925469: (below main) (libc-start.c:185)
==28538==  Address 0x0 is not stack'd, malloc'd or (recently) free'd
==28538== 
==28538== 
==28538== Process terminating with default action of signal 11 (SIGSEGV)
==28538==  Bad permissions for mapped region at address 0x0
==28538==    at 0x0: ???
==28538==    by 0x804B091: native_hwtimer_pre_init (in /home/harter/work/iot-lab/parts/RIOT/examples/hello-world/bin/native/hello-world.elf)
==28538==    by 0x80493E0: startup (in /home/harter/work/iot-lab/parts/RIOT/examples/hello-world/bin/native/hello-world.elf)
==28538==    by 0x804D731: __libc_csu_init (in /home/harter/work/iot-lab/parts/RIOT/examples/hello-world/bin/native/hello-world.elf)
==28538==    by 0x4925469: (below main) (libc-start.c:185)
==28538== 

```

I'm running `Ubuntu 12.04` with gcc `4.6.3`:

```
$ cat /etc/lsb-release 
DISTRIB_ID=Ubuntu
DISTRIB_RELEASE=12.04
DISTRIB_CODENAME=precise
DISTRIB_DESCRIPTION="Ubuntu 12.04.5 LTS"

$ gcc --version
gcc (Ubuntu/Linaro 4.6.3-1ubuntu5) 4.6.3
Copyright (C) 2011 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```

LudwigOrtmann, 2015-03-31:
Toolchain is not supported: https://github.com/RIOT-OS/RIOT/wiki/Family:-native#toolchains

closed 2015-03-31 by authmillenon

cladmi, 2015-04-01:
Ok, I was expecting something like that.

Would it be possible to check for these forbidden versions at compile time?
They are already "hardwritten" in the wiki they could also be in the code.

OlegHahm, 2015-04-01:
@LudwigOrtmann, do you know what is the problem behind this Segfault?

LudwigOrtmann, 2015-04-02:
I think the libc is implementing a system call in an impossible to work around way, but it's been a while since I looked into this.