RIOT-OS/RIOT #2755: Define helper shell commands

OlegHahm, 2015-04-02:
Picking up the discussion from https://github.com/RIOT-OS/RIOT/pull/2583#issuecomment-88922363, I think it would make sense to distinguish between shell commands that helps to request or configure device or network properties and commands that are mostly meant as helpers during development or debugging. Examples for the first type of commands are `ifconfig` (to request network interface properties), `chan` (to set a radio channel) or `rtc` (to query the  datetime) from an RTC. Examples for the last type are the discussed `ncache` (that manipulates IPv6 neighbor cache) and `txtsend` (to send link layer packets).

authmillenon, 2015-04-02:
As suggested in #2583: I think the best approach is to introduce a `shell_helper` module in addition to `shell_commands`. The first type of commands would remain in `shell_commands`, while the second would be moved to `shell_helper`.

PeterKietzmann, 2017-11-27:
@OlegHahm nothing happened in here for the last 2,5 years. Mind to close the issue?

miri64, 2017-11-27:
`txtsnd` went into the direction this PR asked for, `ncache` will be removed with #7925 (and replaced with the more versatile `nib` command), so and overall the momentum seems to go more into the direction to define separate modules for optional helper commands. I would say close as well (@OlegHahm reopen if you disagree).