RIOT-OS/RIOT #2760: drivers/periph/gpio: add output strength option?

LudwigOrtmann, 2015-04-03:
In https://github.com/RIOT-OS/RIOT/pull/2717 the question came up whether it is common for GPIO hardware to support different output levels.
If it is, should we extend the GPIO API in this regard?
If not, how should this possibility be handled?

haukepetersen, 2015-04-03:
What are output levels?

PeterKietzmann, 2015-04-07:
I guess the "Output Driver Strength" was meant, right? On samr21-xpro you can select between "normal" and "sronger" drive strength (compare [datasheet](http://www.atmel.com/Images/Atmel-42223-SAM-R21_Datasheet.pdf) p.398, sec.21.913)
In general this parameter changes the output impemndace of a pin and therefore the maximal output current. Is an option like that available also on other RIOT-cpus/boards? I haven't seen it elsewhere but even didn't look for it. If we find more boards with such an option I'd vote pro API-change. Otherwise I'm more against.

gebart, 2015-04-07:
@PeterKietzmann  The Kinetis family also have configurable drive strength (low = +2/-2 mA, high = +10/-9 mA) on the digital outputs.

kaspar030, 2015-04-07:
I'd say, let's make it optional (e.g., add another function with the extra parameter).

haukepetersen, 2015-04-07:
Generally I am against including this kind of option into the API, as in my experience these options are used rather academically in practice (-> see speed option for STM controllers...). Also these kind of options differ greatly between CPU families and lead quickly to non-portable code. I vote strongly for hiding these options through good default values.

PeterKietzmann, 2015-04-07:
On the one hand I share @haukepetersen's opinion that such an API-change could quickly lead to non-portable code but on the other hand I see the need of a functionality like this by the others (compare  #2717). In general we want to be as energy efficient as possible which should lead to the conclusion that the "default" initialization should happen with the smallest drive strength. If this is the way to go we need a way to increase the drive strength. Introducing an optional function like proposed by @kaspar030 could maybe allow this. But I still see the risk of non-portability. What if sensor driver XY needs a drive strength that is fulfilled by one board with the "lower strength" and on the other board you need to initialize the "higher strength"? Could this limitation be solved by results of #2528? Here we discussed about taking bus initializations outside a sensor driver. 

Well, this was just thinking "loud". Cheers!

gebart, 2015-04-07:
@PeterKietzmann Usually the things that require stronger drive strength is when you drive a LED directly from an output pin or some similar scenario, but I guess there are cases where it could be necessary even to enable stronger drive strength on a communication bus line (when using strong pull-ups on an open drain bus line for example).

haukepetersen, 2015-04-13:
The more I think about it the more I am opposed to include such an option into the API. For CPUs that have this kind of option, I would include it in the `periph_conf`. On the Samd21 I would even tend to put in the higher drive strength as default: remember, just because the pin is able to deliver 7mA, it does not mean that this current is actually delivered (e.g. high impedance device connected to pin)... So I would bet that the actual measurable difference is almost 0?!

PeterKietzmann, 2015-04-13:
Good point about the actual consumtion (difference). This is something we should test. A soultion with Makros in the periph_conf would be sufficient for me.

bapclenet, 2015-04-13:
Well, I thought about it as well and I don't think it's a good idea to add it in the API. Either set on or off, the external component won't consume more than it needs as @haukepetersen mentioned. Let's just try if it really doesn't consume more and then simply set it on on the samd21 every time a pin is set as an output (Then a PR is needed as I mentioned #2717)

PeterKietzmann, 2015-04-14:
In addition: A solution where options are set through the periph_conf can be seen in [this](https://github.com/RIOT-OS/RIOT/pull/2063/files#diff-134bda29ded96fa4abaa9f99216a6116) (WIP) PR

haukepetersen, 2015-04-14:
Jap, thats exactly what the periph_conf is for :-)

bapclenet, 2015-04-15:
@PeterKietzmann, I think it's the best solution! Easy to implement without changing the API

LudwigOrtmann, 2015-04-16:
@haukepetersen @bapclenet @PeterKietzmann @gebart @kaspar030 
So everyone agrees we close this issue?

kaspar030, 2015-04-16:
Let's close it until an actual use case pops up.

PeterKietzmann, 2015-04-16:
Agree

PeterKietzmann, 2015-04-16:
As we found a potential solution and 4/5 of us are pro, 1/5 didn't comment on it, I'll close this issue.

closed 2015-04-16 by PeterKietzmann

RIOT-notifier, 2015-04-16:
Ok

bapclenet, 2015-04-16:
Ok for me, I will deal with periph_conf then.