RIOT-OS/RIOT #2768: drivers/periph: putting the `DEVELHELP` around boundary checks

PeterKietzmann, 2015-04-09:
Should we put the  the `DEVELHELP` around boundary checks in peripheral driver functions? This issue is a place to carry on the discussions in #2701

OlegHahm, 2015-04-09:
On a very first glance: sounds good to me. :+1:

PeterKietzmann, 2015-04-09:
I also think it's beneficial and doesn't hurt anyone.

gebart, 2015-06-25:
should we close this in favor of #3216 ?

gebart, 2015-06-25:
nevermind, there may be more periph drivers that need to be modified.

haukepetersen, 2015-06-26:
I would like to propose the following scheme for the peripheral drivers (as already merged for GPIO and PR'd for SPI -> #3216): We do boundary checks only in the initialization function(s), and change the rest of the access functions to a `void xx(...)` signature.

The rationale behind this is, that once you configured a peripheral, there will be hardly errors (except those that are explicitly defined for that peripheral, e.g. for I2C). If you look at current peripheral driver implementations, the only error they catch is a wrong device given, which is clearly a usage fault but no internal error...

I have also already re-modeled the UART driver to this paradigm, and will PR it soon.

haukepetersen, 2015-06-26:
Ok, in the sense of this discussion I PR's a change to the UART interface to make my point: #3265

PeterKietzmann, 2016-03-18:
@haukepetersen is this addressed  by all your driver refactoring somehow? I can't remember why I opened this issue once, but I don't feel responsible for it :-)

haukepetersen, 2016-03-18:
IMHO this should be solved -> closing it.

closed 2016-03-18 by haukepetersen