RIOT-OS/RIOT #2772: oonf_api requires undocumented define

OlegHahm, 2015-04-09:
Building an application (e.g. https://github.com/Lotterleben/RIOT-AODVv2/tree/master/aodvv2_demo) with `oonf_api` fails with an error like
```
In file included from /home/oleg/git/RIOT/pkg/oonf_api/oonf_api/src-api/common/avl_comp.h:46:0,
                 from /home/oleg/git/RIOT/pkg/oonf_api/oonf_api/src-api/common/avl_comp.c:45:
/home/oleg/git/RIOT/pkg/oonf_api/oonf_api/src-api/common/netaddr.h: In function ‘netaddr_socket_get_addressfamily’
/home/oleg/git/RIOT/pkg/oonf_api/oonf_api/src-api/common/netaddr.h:348:16: error: ‘sockaddr6_t’ has no member named ‘__in6_a’
   return s->std.sin6_family;
```
if one does not define the macro `RIOT` (which is done by the given example in the application Makefile by
`export CFLAGS += -DRIOT`

I think this define is useless, because it's only used in a RIOT specific patch (https://github.com/RIOT-OS/RIOT/blob/master/pkg/oonf_api/0001-add-RIOT-support.patch#L87) that can live without this ifdeff since the package inside RIOT will always be built for RIOT.

If there's still a reason for this define, it should be either documented somewhere and get a better name.

OlegHahm, 2015-04-09:
Thanks to @mfrey for detecting this.

Lotterleben, 2015-04-09:
Seems reasonable to me. @benpicco, since you're the author of this patch, can you tell us if you had any other intentions we're overlooking here?

benpicco, 2015-04-14:
I think the intention was to have this always defined by a core RIOT Makefile, but that never got merged. I wanted to compile the same source code for RIOT and for Linux, so I needed a way to [identify the platform](http://stackoverflow.com/a/4605893) - maybe something like this has been added by now? Otherwise it can't hurt to add such define to `Makefile.include`, better name it `__RIOT__` for consistency.

OlegHahm, 2015-04-14:
I'm not sure if we need this. Is it really a use case to build a thirdparty library inside RIOT (i.e. in `$(RIOTBASE)/pkg/`) for Linux?

benpicco, 2015-04-14:
Well it's possible that one day such patches could be merged upstream, then they mustn't break the build for non-RIOT targets.

OlegHahm, 2015-04-15:
Ok, makes sense. Let's close this as soon as #2805 is merged.

Lotterleben, 2015-04-15:
Will keep an eye on it.

PR #2805 (2015-04-19): oonf_api: use RIOT_VERSION as platform identifier for RIOT
Some third-party code my come with `#ifdef`s to conditionally compile some parts for RIOT, while maintaining the possibility to compile for Linux, Windows, etc.

This makes `oonf_api` check if `RIOT_VERSION` is defined for RIOT specific codepaths. 

fixes #2772
closed 2015-04-19 by OlegHahm