RIOT-OS/RIOT #2794: pktbuf_release problem with ipv6 / icmpv6

kaspar030, 2015-04-12:
I stumbled on a bug that gets triggered when icmpv6 echo request packets arrive very fast (e.g., using ping6 -f).
I'm using the stack as @authmillenon described in his mini-howto from today (just netbase, ipv6, icmpv6 plus driver), using my unpublished encx24j600 driver. I could not reproduce this using the native driver, but I guess the sending is just too fast there.

Somehow a pktsnip with a circular dependency (```pktsnip->next->next == pktsnip```) arrives at nomac, causing the send-function of the driver go crazy (loop until crash).

Slowly arriving packets don't cause a problem.

When packets arrive fast (ping6 -f, the board crashes reliably with the second packet), they queue up before ipv6 can demux them. Raising the priority of ipv6 from PRIORITY_MAIN -3 to PRIORITY_MAIN -4 solves the problem, so does disabling ```pktbuf_release``` (until the memory runs out).

Any ideas?

kaspar030, 2015-04-12:
I was able to reproduce on native.

To trigger this bug, I inserted a long delay (```volatile int i = 100000000; while(i--);```) into the ```_isr_event``` handler.

Same problem, the driver get's a pktsnip with circular pointers.

authmillenon, 2015-04-13:
I'm a little bit confused about were the packet comes Form because in the initial description you say it happens on send while your reproducibility note refers to ist_event (which is called usually in receiving context, unless you are speakimg oft TX started/TX complete events.

For sending I would look into ng_icmpv6_echo_req_handle() since the payload of the request is used for the payload of the reply IIRC. I can imagine thal this might cause a circular pkt list, if for some reason the remove operation of the preceiding headers is skipped... I only am able to test this tomorrow. Can you try and see what happens if you copy the payload into the new packet and remove the request completely?

kaspar030, 2015-04-13:
I think I got it.

At the beginning of ```_ipv6_send_unicast()```, an existing netif header is deleted from the to-send ```pkt``` using ```LL_DELETE()``` and then free'ed using ```pktbuf_release```.

The problem is that ```LL_DELETE()``` doesn't ```NULL``` the ```->next``` pointer of the removed element. If that element now gets freed, the pkt it was part of gets freed, too.

Adding ```netif->next = NULL;``` after ```LL_DELETE()``` solves the issue.

We probably have to check other uses of ```LL_DELETE()```.

authmillenon, 2015-04-13:
We also might fix `LL_DELETE()` ;-)

kaspar030, 2015-04-13:
Sure, but I thought as utlist is a library, not everyone wants this behaviour of a linked list delete... Your call!

Maybe ```static inline pktbuf_remove_and_release(haystack, needle) {..}```?

authmillenon, 2015-04-13:
@haukepetersen what do you think of this?

OlegHahm, 2015-04-13:
I would not touch `LL_DELETE`. utlist is a widely used and known library and having a differing implementation may cause confusion.

authmillenon, 2015-04-13:
I was refering to the new function proposed by @kaspar030.

haukepetersen, 2015-04-13:
I am in favor of @kaspar030s solution. Just thinking loud, how about something like
```c
static inline ng_pktsnip_t *pktbuf_remove_head(ng_pktsnip_t *pkt);
```
?

authmillenon, 2015-04-13:
Since I need to remove stuff from the middle of the list in 6LoWPAN e.g. (IPv6 and UDP headers) this is not general enough for me. I prefer @kaspar030's definition:

static inline ng_pktsnip_t *pktbuf_remove_head(ng_pktsnip_t *pkt, ng_pktsnip_t *snip);

haukepetersen, 2015-04-13:
agreed, then we should not consider my suggestion. Does `LL_DELETE` handle the two extrem cases, where `pkt` is either the first or last snip in the list?

authmillenon, 2015-04-13:
Yes.

authmillenon, 2015-04-13:
(Though last snip is not an extreme case in a single linked list ;-))

authmillenon, 2015-04-14:
See #2804.

closed 2015-05-26 by kaspar030