RIOT-OS/RIOT #2808: Unused paremeter in function ng_netreg_hdr_build

mfrey, 2015-04-14:
There seems to be an unused parameter in file ``sys/net/crosslayer/ng_netreg/ng_netreg.c`` in line 133 at 
``` C
133 ng_pktsnip_t *ng_netreg_hdr_build(ng_nettype_t type, ng_pktsnip_t *payload,
134                                   uint8_t *src, uint8_t src_len,
135                                   uint8_t *dst, uint8_t dst_len)
136 {
```
This was introduced with 69c593b8.

OlegHahm, 2015-04-14:
>  I don't know if that is actually considered an issue.

Warning are always considered an issue.

cgundogan, 2015-04-14:
I would suggest `(void)`'ing the parameters in the `default` clause

authmillenon, 2015-04-14:
The suggested fix was introduced in 716a1bb0 (#2738)

OlegHahm, 2015-04-14:
Eh, the mentioned fix does not comprise `payload`.

authmillenon, 2015-04-14:
Fixed in #2809

authmillenon, 2015-04-14:
Sorry wrong button ^^"

PR #2809 (2015-04-15): netreg: ignore payload on default case in header build
Fixes #2808.
closed 2015-04-15 by OlegHahm