RIOT-OS/RIOT #2838: Doxygen: Order of modules is arbitrary

OlegHahm, 2015-04-21:
The order of the modules on  http://doc.riot-os.org/modules.html is pretty arbitrary (check for instance http://doc.riot-os.org/group__net.html). Looking at doxygen configuration and documentation. There are basically two ways to deal with that:
1. Set `SORT_GROUP_NAMES` to `YES` and let doxygen order stuff alphabetically.
2. Create all groups at central point (directly after `doc/doxygen/mainpage`) and define the order there at a central point.

While I'm not sure that alphabetic ordering is the most meaningful, I think it would be still better than more or less no ordering.

phiros, 2015-04-23:
I think using the alphabetical ordering is a good idea for now. The other option would essentially add yet another file that has to be edited when adding / changing a module.

OlegHahm, 2015-04-23:
...ad could lead to discussion about the "best" ordering. I tend to agree.

x3ro, 2015-04-23:
> While I'm not sure that alphabetic ordering is the most meaningful, I think it would be still better than more or less no ordering. @OlegHahm

> I think using the alphabetical ordering is a good idea for now. @phiros 

Completely agree.

OlegHahm, 2015-04-23:
And here we go: #2859.

phiros, 2015-04-26:
Fixed in -> #2859

closed 2015-04-26 by phiros