RIOT-OS/RIOT #2840: Failing `ng_netreg_getnext` unit test

x3ro, 2015-04-21:
I've been seeing this one fail a couple of times and wanted to inquire whether or not a fix is one the way :smile: Happens to me on current `HEAD` (0d45479)

    netreg_tests.test_netreg_getnext__2_entries (/home/lucas/RIOT/tests/unittests/tests-netreg/tests-netreg.c 150) ng_netreg_getnext(entry) was null.

authmillenon, 2015-04-21:
I've never encountered this, but will test.

authmillenon, 2015-04-21:
```
[martine@beutlin RIOT]<3 git show --oneline
0d45479 Merge pull request #2819 from kushalsingh007/hash_string_missing

[martine@beutlin RIOT]<3 make -C tests/unittests/ all test
make: Entering directory `/home/martine/Repositories/RIOT-OS/RIOT/tests/unittests'
Building application "unittests" for "native" with MCU "native".

touch /home/martine/Repositories/RIOT-OS/RIOT/tests/unittests/main.c
"make" -C /home/martine/Repositories/RIOT-OS/RIOT/boards/native
"make" -C /home/martine/Repositories/RIOT-OS/RIOT/boards/native/drivers
"make" -C /home/martine/Repositories/RIOT-OS/RIOT/core
"make" -C /home/martine/Repositories/RIOT-OS/RIOT/cpu/native
"make" -C /home/martine/Repositories/RIOT-OS/RIOT/cpu/native/periph
"make" -C /home/martine/Repositories/RIOT-OS/RIOT/drivers
"make" -C /home/martine/Repositories/RIOT-OS/RIOT/drivers/netdev/base
"make" -C /home/martine/Repositories/RIOT-OS/RIOT/sys
"make" -C /home/martine/Repositories/RIOT-OS/RIOT/sys/base64
"make" -C /home/martine/Repositories/RIOT-OS/RIOT/sys/bloom
"make" -C /home/martine/Repositories/RIOT-OS/RIOT/sys/cbor
"make" -C /home/martine/Repositories/RIOT-OS/RIOT/sys/crypto
"make" -C /home/martine/Repositories/RIOT-OS/RIOT/sys/embunit
"make" -C /home/martine/Repositories/RIOT-OS/RIOT/sys/hash_string
"make" -C /home/martine/Repositories/RIOT-OS/RIOT/sys/hashes
"make" -C /home/martine/Repositories/RIOT-OS/RIOT/sys/net/crosslayer/net_help
"make" -C /home/martine/Repositories/RIOT-OS/RIOT/sys/net/crosslayer/ng_inet_csum
"make" -C /home/martine/Repositories/RIOT-OS/RIOT/sys/net/crosslayer/ng_netif
"make" -C /home/martine/Repositories/RIOT-OS/RIOT/sys/net/crosslayer/ng_netreg
"make" -C /home/martine/Repositories/RIOT-OS/RIOT/sys/net/crosslayer/ng_pktbuf
"make" -C /home/martine/Repositories/RIOT-OS/RIOT/sys/net/network_layer/fib
"make" -C /home/martine/Repositories/RIOT-OS/RIOT/sys/net/network_layer/ng_ipv6/addr
"make" -C /home/martine/Repositories/RIOT-OS/RIOT/sys/net/network_layer/ng_ipv6/hdr
"make" -C /home/martine/Repositories/RIOT-OS/RIOT/sys/net/network_layer/ng_ipv6/nc
"make" -C /home/martine/Repositories/RIOT-OS/RIOT/sys/net/network_layer/ng_ipv6/netif
"make" -C /home/martine/Repositories/RIOT-OS/RIOT/sys/net/network_layer/ng_sixlowpan/ctx
"make" -C /home/martine/Repositories/RIOT-OS/RIOT/sys/pipe
"make" -C /home/martine/Repositories/RIOT-OS/RIOT/sys/timex
"make" -C /home/martine/Repositories/RIOT-OS/RIOT/sys/ubjson
"make" -C /home/martine/Repositories/RIOT-OS/RIOT/sys/vtimer
"make" -C /home/martine/Repositories/RIOT-OS/RIOT/tests/unittests/netdev_dummy
"make" -C /home/martine/Repositories/RIOT-OS/RIOT/tests/unittests/tests-base64
"make" -C /home/martine/Repositories/RIOT-OS/RIOT/tests/unittests/tests-bloom
"make" -C /home/martine/Repositories/RIOT-OS/RIOT/tests/unittests/tests-cbor
"make" -C /home/martine/Repositories/RIOT-OS/RIOT/tests/unittests/tests-core
"make" -C /home/martine/Repositories/RIOT-OS/RIOT/tests/unittests/tests-crypto
"make" -C /home/martine/Repositories/RIOT-OS/RIOT/tests/unittests/tests-fib
"make" -C /home/martine/Repositories/RIOT-OS/RIOT/tests/unittests/tests-hash_string
"make" -C /home/martine/Repositories/RIOT-OS/RIOT/tests/unittests/tests-inet_csum
"make" -C /home/martine/Repositories/RIOT-OS/RIOT/tests/unittests/tests-ipv6_addr
"make" -C /home/martine/Repositories/RIOT-OS/RIOT/tests/unittests/tests-ipv6_hdr
"make" -C /home/martine/Repositories/RIOT-OS/RIOT/tests/unittests/tests-ipv6_nc
"make" -C /home/martine/Repositories/RIOT-OS/RIOT/tests/unittests/tests-ipv6_netif
"make" -C /home/martine/Repositories/RIOT-OS/RIOT/tests/unittests/tests-netdev_dummy
"make" -C /home/martine/Repositories/RIOT-OS/RIOT/tests/unittests/tests-netif
"make" -C /home/martine/Repositories/RIOT-OS/RIOT/tests/unittests/tests-netreg
"make" -C /home/martine/Repositories/RIOT-OS/RIOT/tests/unittests/tests-pkt
"make" -C /home/martine/Repositories/RIOT-OS/RIOT/tests/unittests/tests-pktbuf
"make" -C /home/martine/Repositories/RIOT-OS/RIOT/tests/unittests/tests-pktqueue
"make" -C /home/martine/Repositories/RIOT-OS/RIOT/tests/unittests/tests-sixlowpan_ctx
"make" -C /home/martine/Repositories/RIOT-OS/RIOT/tests/unittests/tests-timex
"make" -C /home/martine/Repositories/RIOT-OS/RIOT/tests/unittests/tests-ubjson
   text	   data	    bss	    dec	    hex	filename
 345827	    608	  79316	 425751	  67f17	/home/martine/Repositories/RIOT-OS/RIOT/tests/unittests/bin/native/unittests.elf
RIOT native interrupts/signals initialized.
LED_GREEN_OFF
LED_RED_ON
RIOT native board initialized.
RIOT native hardware initialization complete.

kernel_init(): This is RIOT! (Version: 2014.12-1033-g9c2663-ng_sixlowpan_ctx/feat/initial)
kernel_init(): jumping into first task...
........................................................................................................................................................................................................................................................................................................................................................................................................................................
OK (424 tests)
XXX: lpm_set(): LPM_POWERDOWN not implemented
lpm_set(): exit()
make: Leaving directory `/home/martine/Repositories/RIOT-OS/RIOT/tests/unittests'
```

I've got no error here.

x3ro, 2015-04-21:
Doesn't seem to occur when building with clang (3.5.0-4ubuntu2), but does occur when building with `gcc version 4.9.1 (Ubuntu 4.9.1-16ubuntu6)`.

authmillenon, 2015-04-21:
Okay, I've got `gcc (Ubuntu 4.8.2-19ubuntu1) 4.8.2`

x3ro, 2015-04-21:
Checked and doesn't happen with `gcc 4.8.3-12ubuntu3`. So far only 4.9 seems to be affected.

authmillenon, 2015-04-21:
Can you step through with gdb in your version and see what happens?

```gdb
b test_netreg_getnext__2_entries
run
print netreg
n
n
print netreg
```

should be the most interesting (maybe step into `ng_netreg_lookup()` and `ng_netreg_getnext()` afterwards to see what `getnext` does)

x3ro, 2015-04-21:
## Take 1

    Breakpoint 1, test_netreg_getnext__2_entries () at /home/lucas/RIOT/tests/unittests/tests-netreg/tests-netreg.c:146
    146     ng_netreg_entry_t *entry = NULL;
    (gdb) print netreg
    $1 = {0x0, 0x0}
    (gdb) n
    148     test_netreg_num__2_entries();
    (gdb) n
    149     TEST_ASSERT_NOT_NULL((entry = ng_netreg_lookup(NG_NETTYPE_TEST, TEST_UINT16)));
    (gdb) print netreg
    $2 = {0x0, 0x809ef80 <main_stack+15680>}
    (gdb)

## Take 2 (stepping into lookup/getnext)

Breakpoint 1, test_netreg_getnext__2_entries () at /home/lucas/RIOT/tests/unittests/tests-netreg/tests-netreg.c:146
146     ng_netreg_entry_t *entry = NULL;
(gdb) n
148     test_netreg_num__2_entries();
(gdb) n
149     TEST_ASSERT_NOT_NULL((entry = ng_netreg_lookup(NG_NETTYPE_TEST, TEST_UINT16)));
(gdb) s
ng_netreg_lookup (type=NG_NETTYPE_TEST, demux_ctx=10098) at /home/lucas/RIOT/sys/net/crosslayer/ng_netreg/ng_netreg.c:60
60      if (_INVALID_TYPE(type)) {
(gdb) p type
$1 = NG_NETTYPE_TEST
(gdb) n
64      LL_SEARCH_SCALAR(netreg[type], res, demux_ctx, demux_ctx);
(gdb) n
66      return res;
(gdb) p res
$2 = (ng_netreg_entry_t *) 0x809ef74 <main_stack+15668>
(gdb) n
67  }
(gdb) n
test_netreg_getnext__2_entries () at /home/lucas/RIOT/tests/unittests/tests-netreg/tests-netreg.c:150
150     TEST_ASSERT_NOT_NULL(ng_netreg_getnext(entry));
(gdb) s
ng_netreg_getnext (entry=0x809ef74 <main_stack+15668>) at /home/lucas/RIOT/sys/net/crosslayer/ng_netreg/ng_netreg.c:95
95      if (entry == NULL) {
(gdb) p entry
$3 = (ng_netreg_entry_t *) 0x809ef74 <main_stack+15668>
(gdb) n
99      demux_ctx = entry->demux_ctx;
(gdb) p demux_ctx
$4 = 134868852
(gdb) n
101     LL_SEARCH_SCALAR(entry->next, entry, demux_ctx, demux_ctx);
(gdb) n
103     return entry;
(gdb) p entry
$5 = (ng_netreg_entry_t *) 0x0
(gdb)

authmillenon, 2015-04-21:
Can you do Take 1 again, this time print `netreg[NG_NETTYPE_TEST]->next`, `netreg[NG_NETTYPE_TEST]->next->next`, … (until it hits `NULL`) before calling `ng_netreg_lookup(NG_NETTYPE_TEST, TEST_UINT16)` and after? (optionally step in, if something changes).

If nothing changes, please have a closer look at `test_netreg_num__2_entries()`...

x3ro, 2015-04-21:
Like this?

```
Breakpoint 1, test_netreg_getnext__2_entries () at /home/lucas/RIOT/tests/unittests/tests-netreg/tests-netreg.c:146
146	    ng_netreg_entry_t *entry = NULL;
(gdb) n
148	    test_netreg_num__2_entries();
(gdb) n
149	    TEST_ASSERT_NOT_NULL((entry = ng_netreg_lookup(NG_NETTYPE_TEST, TEST_UINT16)));
(gdb) p netreg[NG_NETTYPE_TEST]->next
$6 = (struct ng_netreg_entry *) 0x809ef74 <main_stack+15668>
(gdb) p netreg[NG_NETTYPE_TEST]->next->next
$7 = (struct ng_netreg_entry *) 0x0
(gdb) n
150	    TEST_ASSERT_NOT_NULL(ng_netreg_getnext(entry));
(gdb) p netreg[NG_NETTYPE_TEST]->next
$8 = (struct ng_netreg_entry *) 0x809ef74 <main_stack+15668>
(gdb) p netreg[NG_NETTYPE_TEST]->next->next
$9 = (struct ng_netreg_entry *) 0x0
```

authmillenon, 2015-04-21:
Ahhhhhh I think know what happens …

authmillenon, 2015-04-21:
(The stack gets overwritten)

authmillenon, 2015-04-21:
@x3ro Can you test if it still fails with #2841?

x3ro, 2015-04-21:
Fixed by #2841

authmillenon, 2015-04-21:
Is not merged yet.

PR #2841 (2015-04-21): unittests: move netreg entries out of stack space
Fixes #2840.
closed 2015-04-21 by authmillenon