RIOT-OS/RIOT #2851: msg: Allocating the message queue

OlegHahm, 2015-04-22:
Follow up to this discussion: https://github.com/RIOT-OS/RIOT/pull/2454/files#r28869027

If a thread uses a [message queue](http://doc.riot-os.org/group__core__msg.html#ga90d8645ba6c1eca6ace05e369b07ce5b) for incoming messages it has to allocate the memory for this queue somewhere. I always used a global static array for this purpose, but just realized that several modules (mostly of the network stack) allocate this memory directly on the stack. I wonder if this a good idea, since a developer may change the size of a message queue without knowing that this affects the stack size. Furthermore, this makes it either impossible to define aggressively optimized default stack sizes or forces developers to define their own stack sizes if the use the message queue. (Actually, this could be done pretty easy by using `KERNEL_CONF_STACKSIZE_DEFAULT + sizeof(msg_q)`.)

In general, I feel that this is something where we should agree on a typical way to handle it and add it to RIOT's best current practice guidelines (tbd).

kaspar030, 2015-04-22:
IMHO allocating anything larger than <verysmall> should not be allocated on the stack. So in general that applies for message queues.

+1 for declaring definition of message queues on the stack for bad practice.

Kijewski, 2015-04-22:
Nb. I think that it would be more sane if the message queue would get allocated in `thread_create()`, but I daresay that you don't want to add further arguments to `thread_create()` … :)

authmillenon, 2015-04-22:
> IMHO allocating anything larger than should not be allocated on the stack. So in general that applies for message queues.

larger than what?

OlegHahm, 2015-04-22:
The notification mails said:
`IMHO allocating anything larger than <verysmall> should not be allocated on the stack. So in general that applies for message queues.`

authmillenon, 2015-04-22:
Okay, added the backslashes (md interprets `<bla>`s as HTML tags)

OlegHahm, 2015-05-18:
So, we agreed that this should go into the RIOT BPP, right?

haukepetersen, 2015-07-02:
I re-opened this issue, as I am not convinced of the outcome - actually I would argue for putting message queues on the stack! Some modules can be instantiated multiple times during runtime. There are scenarios where we don't even now the actual number beforehand. In these cases, I think it makes much sense to have the message queue on the stack, is it prevents having to manage some extra memory locations during runtime...

OlegHahm, 2015-07-02:
I disagree: you have to know the number of instances at compile time in order to allocate the stack for their threads anyway.

kaspar030, 2015-11-28:
Me too. We should keep stacks small. 

@haukepetersen What's your scenario where you want to instantiate module instances at runtime?

OlegHahm, 2016-05-23:
This is still an issue. Just had a crashing node because I increased the message queue size for `gnrc_netdev2`.

kaspar030, 2016-11-18:
We could add a DEVELHELP'ed assertion that, should the msg array be between stack pointer and stack size, checks if there's a certain amount of stack space left.

haukepetersen, 2016-11-18:
Actually, after all this time, I don't have an opinion about this anymore -> do whatever you want :-)

miri64, 2016-11-18:
> We could add a DEVELHELP'ed assertion that, should the msg array be between stack pointer and stack size, checks if there's a certain amount of stack space left.

Assert is already only activated, when DEVELHELP is defined ;-). But we should be careful about the error message. Assertions already are confusing many people.

OlegHahm, 2016-11-18:
> Assert is already only activated, when DEVELHELP is defined ;-).

Or `FORCE_ASSERTS` is set to 1.