RIOT-OS/RIOT #2853: ng_pktbuf: documentation is confusing

OlegHahm, 2015-04-22:
I just tried to understand what happens if I call `ng_pktbuf_add()` when passing the payload of a L2 frame as a first argument. However, the doxygen (particular from [line 72](https://github.com/RIOT-OS/RIOT/blob/master/sys/include/net/ng_pktbuf.h#L72) to 91) are pretty confusing to me. I do not even understand what `result` is referring to.

authmillenon, 2015-04-22:
The result is the return value. What else is confusing to you? I tried to be as verbose (but not *too* verbose) as possible in the documentation.

OlegHahm, 2015-04-22:
Aha, that already helps a lot, now the former rather cryptic `ng_pktsnip_t::next of result = pkt->next` makes sense. Maybe that's obvious for some people (apparently not for me), but I think we should make this explicit: The return value of this function is a `ng_pktsnip_t` struct referred to as `result`.

I also do not quite understand the graphic.

All in all, I understand you correctly: when you pass an existing pkgsnip_t to this function (plus some extra conditions are fulfilled) the packet is assumed as receiving and an encapsulated header can be marked. Right?

And to what pktsnip does `@param[in] type      Protocol type of the ng_pktsnip_t.` refer to?

authmillenon, 2015-04-22:
> All in all, I understand you correctly: when you pass an existing pkgsnip_t to this function (plus some extra conditions are fulfilled) the packet is assumed as receiving and an encapsulated header can be marked. Right?

the extra conditions and not the fact that a given pktsnip_t is passed are what indicate that it is receiving (otherwise it is assumed as sending), but if you meant that, you are correct.

> And to what pktsnip does @param[in] type Protocol type of the ng_pktsnip_t. refer to?

The result.

authmillenon, 2015-04-22:
> I also do not quite understand the graphic.

Maybe I can help to make things clearer.

authmillenon, 2015-04-22:
See #2854

PR #2854 (2015-04-22): pktbuf: clarify doc of pktbuf_add()
Fixes #2853
commit d994908d (2015-04-22): pktbuf: clarify doc of pktbuf_add()

Fixes #2853
closed 2015-04-22 by authmillenon