RIOT-OS/RIOT #2860: ng_pktbuf: about pktbuf_start_write

OlegHahm, 2015-04-23:
While reviewing the initial [IPv6 import](#2454) I came several times across the function [ng_pktbuf_start_write](https://github.com/RIOT-OS/RIOT/blob/master/sys/include/net/ng_pktbuf.h#L173). After thinking (admittedly short) about this function I have two questions:
1. Basically the pktbuf works on semaphores. While `_release()` and `_hold()` are pretty self-explanatory function names in this context, `_start_write()` is less speaking and sounds a bit clumsy. Wouldn't either something like `_get_write_lock()` or `_duplicate()` be more to the point?
2. Is this duplication really necessary? Could an operation not just block until the other thread is done with the chunk? I'm a little bit worried about the memory consumption of the current solution.

authmillenon, 2015-04-23:
Since this function was originally @haukepetersen's idea I reassign.

OlegHahm, 2015-09-03:
@haukepetersen, ping!

haukepetersen, 2016-04-01:
sorry, this dropped out of my field of view as it seems...

@OlegHahm: do your questions still need answers?

In short:
1. the name `start_write` was not chosen for a particular reason. Looking at it today, I would also be in favor of renaming, maybe `lock()` or `write_lock()`? `_duplicate()` is not a good idea, as this does not reflect what is actually happening (the packet is only copied if the user counter is > 1...).

2. Yes, it is necessary.