RIOT-OS/RIOT #2863: cpu: syscall implementation seem to have trouble if uart0 is not enabled

haukepetersen, 2015-04-23:
I observed the same behavior now on different boards (e.g. 'airfy-beacon' or 'iot-lab_M3'): Applications using the shell are running fine if  the `uart0` module is enabled (is in the default app). But when running an application using the shell without `uart0` by reading input chars directly via the clibs `getchar()`, the boards do not react.

Now interestingly, one you put a short output in the `_read_r()` function in the 'CPU/syscalls.c` file, all inputted characters are read to the shell just fine. 

I am not sure where this behavior comes from, my current guess is, that there is a concurrency problem when accessing the ringbuffer from both thread context (the shell thread) and the uart interrupt routine. Any ideas?

PeterKietzmann, 2015-04-24:
Hope to look at it soon

kaspar030, 2015-04-24:
@PeterKietzmann Maybe this is a good opportunity to work on top of #2852 as it unifies ```syscalls.c```?

PeterKietzmann, 2015-04-24:
Good point! Will use it then

PeterKietzmann, 2015-07-30:
@haukepetersen I totally forgot this issue. Is it still current after all the UART refactoring?

kaspar030, 2015-07-30:
Well, I'm using shell with just getchar/putchar on a lot of boards, so I'd say no... Maybe this was the first incarnation of that pendSV bug?

haukepetersen, 2015-07-30:
Yap, I think this should be obsolete after we fixed the pendSV bug. But let me test with an NRF platform once more before we close this.

OlegHahm, 2015-09-02:
I guess this can be closed, right?

haukepetersen, 2015-09-03:
yes, this is obsolete

closed 2015-09-03 by haukepetersen