RIOT-OS/RIOT #2876: drivers/at86rf231 (the old one): subsequent sending packets deadlocks the driver/device

BytesGalore, 2015-04-28:
When sending a number of subsequent packets, the driver/device seem to deadlock in its internal state-machine.
The problem appears  on `iot-lab_M3` nodes for instance when sending packets that need to be fragmented [1]. This way at least 2 subsequent packets are prepared to be sent deadlocking the driver/device.
It also happens if no fragmentation is needed but a number of packets are sent subsequently, for instance when sending UDP packets over a socket in a loop.

As workaround a sufficient large timeout (~10-20ms) can be set in `transceiver.c` whenever  packet needs to be sent before calling the `msg_reply(&m, &m);` [2].

[1] https://github.com/RIOT-OS/RIOT/blob/master/sys/net/network_layer/sixlowpan/lowpan.c#L241
[2] https://github.com/RIOT-OS/RIOT/blob/master/sys/transceiver/transceiver.c#L310

bapclenet, 2015-04-29:
Yeah, those are well-know issues. The at86rf2xx driver has been merged yesterday, you can find a example/test in tests/driver_at86rf2xx. You should use the new one from now.
Let us know if you observe the same behaviour with new driver.

PeterKietzmann, 2015-05-13:
@BytesGalore now we have this issue formulated for the archive, thanks for that. After spending some more time on the old and new network-stack I really think that nobody is going to spend further work on the old stack and therefore on the old driver implementation. So I'll close this issue. Re-open if you don't agree :-)

closed 2015-05-13 by PeterKietzmann