RIOT-OS/RIOT #2887: ng_netif: Erase addr while checking the length

bapclenet, 2015-04-29:
In ``sys/net/crosslayer/ng_netif/ng_netif_addr_from_str.c``, function ``size_t ng_netif_addr_from_str(uint8_t *out, size_t out_len, char *str)``
By rewriting the address with``*str = '\0';`` each time ``(*str == ':') || (*str == '\0')``, the address is changed then.

Ex: 
addr = "18:18"
After the function, addr = "18"

If we want to use the same address as the previous one, it won't work as it has been rewritten.  
Shouldn't we simply copy str in a temp variable?

authmillenon, 2015-04-29:
I don't understand the problem. Can you rephrase please?

authmillenon, 2015-04-29:
Ah now I get it…

no: I would prefer to keep this function as slim as possible. If you need the string afterwards you have to put them in a temporary variable before-hand yourself.

bapclenet, 2015-04-29:
Yep, what I did afterwards. Sorry for the disruption

closed 2015-04-29 by bapclenet

Kijewski, 2015-04-29:
Wouldn't it make more sense to fix the function? (untested)
```c
#include <stdio.h>
#include <stdint.h>

static int _dehex(char c)
{
    return ('0' <= c && c <= '9') ? c - '9'
         : ('A' <= c && c <= 'F') ? c - 'A' + 10
         : ('a' <= c && c <= 'f') ? c - 'a' + 10
         : -1;
}

size_t ng_netif_addr_from_str(uint8_t *out, size_t out_len, const char *str)
{
    size_t len;
    for (len = 0; *str; ++len) {
        if (len >= out_len) {
            return 0;
        }

        int a, b;
        if (((a = _dehex(*str++)) < 0) || ((b = _dehex(*str++)) < 0) || (*str++ != ':')) {
            return 0;
        }

        *out++ = (a << 4) | b;
    }

    return len;
}
```

authmillenon, 2015-04-29:
Go on, open a PR and we will see ;-)

Kijewski, 2015-04-29:
→ #2889