RIOT-OS/RIOT #2888: how can i implement 6lowpan with MSB-A2 temperature sensor nodes please help

saleemrazam, 2015-04-29:


OlegHahm, 2015-04-29:
You basically need an application that includes the following modules:
* defaulttransceiver
* sixlowpan (or udp)
* sht11

The [Wiki](https://github.com/RIOT-OS/RIOT/wiki) includes basic documentation how to set things up, particular [this how-to](https://github.com/RIOT-OS/RIOT/wiki/Creating-your-first-RIOT-project) should contain the most relevant information for you how to get started.

closed 2015-04-29 by OlegHahm

saleemrazam, 2015-04-30:
Thanks OlegHahm.... But there is also an example application in the Examples folder that i have downloaded from the gihub and installed on my ubuntu desktop, i can run the other applications like default, hello-world etc but how can i atcually run that sixlowapp from there..

OlegHahm, 2015-04-30:
You mean https://github.com/RIOT-OS/applications/tree/master/sixlowapp?

saleemrazam, 2015-05-04:
yes,,  i mean this,,, I am trying to flash sixlowapp but it says make: *** No rule to make target `/home/saleemrazam/RIOT/applications/sixlowapp/../../RIOT/Makefile.include'.  Stop.

OlegHahm, 2015-05-04:
Then the path is wrong. The `applications` repository expects to be located at the same folder level as `RIOT` itself. It seems that you checked out `applications` inside `RIOT`. In this case you need to remove change `RIOTBASE ?= $(CURDIR)/../../RIOT` to `RIOTBASE ?= $(CURDIR)/../..`.

saleemrazam, 2015-05-04:
solved,,,,,,,,,,,,,,, lots of thanks..

 cheers