RIOT-OS/RIOT #2894: BOARD_INSUFFICIENT_RAM not respected when building with "make all"

x3ro, 2015-04-30:
When trying to reproduce build issues on Travis, I built the respective boards inside docker and noticed that they were failed because of RAM overflow. After adding the boards to the `BOARD_INSUFFICIENT_RAM` variable of the unittests makefile, the still occurred when building with `make all`, since `RIOTNOLINK` only seems to be set when running `make buildtest`.

Kijewski, 2015-04-30:
When you explicitly supply a BOARD= override, then the project should be built. Maybe an error should be printed as in FEATURES_REQUIRED.

LudwigOrtmann, 2015-05-01:
`s/error/warning`

x3ro, 2017-11-28:
Seems like nobody has run into this and I cannot reproduce it anymore, so I'm closing this. Feel free to reopen in case you encounter this issue.

closed 2017-11-28 by x3ro