RIOT-OS/RIOT #2900: network auto-initialization suboptimal

kaspar030, 2015-05-01:
Hauke, I'm sorry, but I flaws in the new network stack auto initialization.

You basically put network stack initialization into the boards's directory.
Already you can see that every now supported board has basically the same ```auto_init_ng_netif()``` function, only differing in the actual driver initialization function call.

This will lead to massive code duplication.
Just check the diff between the now-supported devices:
```diff boards/iot-lab_M3/auto_init_ng_netif/netif_board.c boards/samr21-xpro/auto_init_ng_netif/netif_board.c```.
They are minimal, more impressive is the shared code.

Want to change the nomac init call? Go change every supported driver.

Your scheme:

(board specific:)
1. ```auto_init_ng_netif()``` sets up devices available on the board and
2. starts the corresponding nomac thread.

I propose a different scheme:

(board specific):
The board makefile sets up ```USEMODULE += device_module``` and the corresponding defines for used pins.
(generic:)
1. there's a ```const static``` list of netdev devices somewhere in autoinit_netdev.c
2. on USEMODULE += driver_module, that list gets entries for the available devices
3. ```auto_init_ng_netif()``` calls the initialization function for used devices and
4. starts the corresponding nomac thread.

That way, in every board/application, only ```USEMODULE += device_driver``` (and maybe the PIN defines) have to be set up, but starting of any thread is generic.

We *need* to keep platform specific code to a minimum. One developer can easily test generic code, but it's impossible to maintain code on different platforms if those platforms are not at hand.

kaspar030, 2015-05-01:
I'll PR a proposal.

OlegHahm, 2015-05-18:
#2901 is merged

OlegHahm, 2015-08-17:
@kaspar030, I assume you're okay.

closed 2015-08-17 by OlegHahm