RIOT-OS/RIOT #2919: at86rf2xx: driver receives interrupts for every message

bapclenet, 2015-05-05:
Concerning the driver at86rf2xx, I don't know if it is an issue or not but the driver module (at86rf2xx) receives an interrupt from the transceiver every single message even if the message is not addressed to the board.
(to see it, enable debug in ng_at86rf2xx_netdev.c, see ``[ng_at86rf2xx] EVT - RX_START`` many times)

I though that the transceiver was able to deal with messages that are not addressed to it and wakes the proc up only for messages which are addressed to the board. 

Maybe I misunderstood something here.

bapclenet, 2015-05-05:
By the way, why do we receive 4 ``[ng_at86rf2xx] EVT - RX_START`` for each message?

OlegHahm, 2015-05-05:
> By the way, why do we receive 4 [ng_at86rf2xx] EVT - RX_START for each message?

IIRC what @haukepetersen said the transceiver retransmit unicast packets up to four times if no ACK is received.

OlegHahm, 2015-05-05:
Looking at the datasheet I think we should disable `IRQ_2` (`RX_START`) and only activate `IRQ_3` (`RX_END`) in normal operation. Since `IRQ_2` is fired as soon as the SFD is received, address recognition is not feasible at this point.

OlegHahm, 2015-05-05:
Can you check with #2920?

bapclenet, 2015-05-06:
Great! I think it's a very good idea. The software doesn't need to be waking up as soon as a message is received (if it's not for it)

bapclenet, 2015-05-07:
#2920 solved the problem, I close the issue

closed 2015-05-07 by bapclenet