RIOT-OS/RIOT #2924: API: Driver RTT - frequency

bapclenet, 2015-05-06:
I'm wondering if RTT driver should include a param for frequency? Otherwise each board defines is own  speed. Should we include a param frequency in ``rtt_init()`` or should the board define this param by itself (as it is now)?

kaspar030, 2015-05-06:
I wonder how much the rtt API will end up different from the timer API. Maybe we can unify them? The basic interface of an RTT will be "initialize", "get current counter value" and "set an alarm callback at value X", which is exactly what a timer does...

haukepetersen, 2015-05-06:
> In my mind, I think that a Real Time Timer has to run at 1Hz by default.

Disagree. It should be up to the platform to define this speed (as not every board is able to run the RTT with exactly 1Hz). In my opinion it should be then up to a more high-level timer module to map the corresponding timer/RTT speeds to physical values.

bapclenet, 2015-05-06:
Thanks for the answer
Ok @haukepetersen, let's say that each RTT defines its speed as the API is already

The problem:
But what could be a default value? (for the samr21 RTT for instance)

haukepetersen, 2015-05-07:
as default value 1Hz seems fine to me.

kaspar030, 2015-05-07:
Well, if we use 1hz as default, and use RTTs to go to very low power modes, then the granularity of those switches will be one second. Do you think that is enough?

bapclenet, 2015-05-07:
4Hz should be fine then? 
Anyway, people will change it in the driver if they want to...

bapclenet, 2015-05-08:
I close this and people will choose according to the board possibilities.

closed 2015-05-08 by bapclenet