RIOT-OS/RIOT #2931: DEVELHELP clumsy, why not use assert()?

kaspar030, 2015-05-08:
While reading in on the discussion on whether to check for parameter validity, I remembered assert.

Why don't we use that? It is ANSI C, it optimizes out cleanly, it is only one small line per check...

e.g.

<pre>
#include <assert.h>

int function(char*buf, int len)
{
    assert(len>minlen);

// blah
}
</pre>

OlegHahm, 2015-05-08:
Sounds good. In a discussion with @mfrey recently I wondered why we were not using assert(), but then forgot it again.

kaspar030, 2015-05-08:
I talked to Hauke about this.
One drawback is that errors are just detected, and cannot be acted upon.

But then again, IMHO we cannot afford to check all return codes for all the little setters and getters we have anyways.

OlegHahm, 2015-05-08:
There's no reason to replace all `DEVELHELP` occurrences with `assert`, but in many places (like the discussed length parameter check) it satisfies our needs. In many cases you cannot really act in a sensible manner upon an error any way on deeply embedded systems apart from restarting.

RIOT-notifier, 2015-05-08:
I remember that I worked with assert a while back, but there were some
problems (I think it was that msp430 and avr don't include it into their
respective libcs).

kaspar030, 2015-05-08:
We could add the header for those platforms. The actual definition can be *very* short.

haukepetersen, 2015-05-08:
> There's no reason to replace all DEVELHELP occurrences with `assert`

+1 (and I think that was exactly the idea here).

For implementing this, I think I would tend to create a RIOT specific assert header with an `ASSERT()` define, and map this then to the clibs assert. On platforms, where assert is not availble in the clib, we can then just implement our own thing...

OlegHahm, 2015-05-08:
Do you think that's necessary? In the it just needs some way to print and call `abort()` (which can be easily mapped to `core_crash()`). The missing definition for MSP430 was merged some time ago IIRC.

haukepetersen, 2015-05-08:
I have no idea. But in the end I do not care how we implement it...

kaspar030, 2015-05-08:
I propose, we add "assert.h" includes to the cpu directories that need it, and those just include "riot_assert.h". For the other platforms, we go with the libc version.

riot_assert.h static inline defines assert() to what the libc implementations do: print something like "assert failed in __func__ __LINE__ ...".

kaspar030, 2015-05-08:
If we make it crash, we cannot see how the failed assert actually crashes the system, and also the system would behave differently depending on asserts being enabled (compiled without NDEBUG).

kaspar030, 2015-05-08:
ah, on native, assert crashes.

kaspar030, 2015-05-08:
and newlib calls "exit(1)", reseting the board.

OlegHahm, 2015-05-08:
Sorry, I meant `core_panic()`.

kaspar030, 2015-11-28:
We're using assert all over the place, so I'll close this as solved.

closed 2015-11-28 by kaspar030