RIOT-OS/RIOT #2961: ng_at86rf2xx: Source address of IPv6 packets get overwritten

authmillenon, 2015-05-10:
Here is a strange bug: after fixing #2959 the source address of an ICMPv6 echo request gets overwritten *after* it was copied into the transmission buffer of the `ng_at86rf2xx` (checked it with GDB).

I compiled `tests/driver_at86rf2xx` with `ng_icmpv6_echo` and `ng_sixlowpan_frag` on branch iotlab_tmp2 in @OlegHahm (commit https://github.com/OlegHahm/RIOT/commit/3b41d70f520e04e64bf974ef72ca5fa229adf9c0) (merged in #2959 of course):

```
ncache
2015-05-10 14:10:48,072 - INFO # > ncache
2015-05-10 14:10:48,102 - INFO # 
2015-05-10 14:10:48,128 - INFO # IPv6 address                    if  L2 address                state      type
2015-05-10 14:10:48,154 - INFO # -----------------------------------------------------------------------------
2015-05-10 14:10:48,175 - INFO # fe80::ff:fe00:e584               7  e5:84                     UNMANAGED   -
ifconfig
2015-05-10 14:10:51,680 - INFO # > ifconfig
2015-05-10 14:10:51,681 - INFO # 
2015-05-10 14:10:51,691 - INFO # Iface  7   HWaddr: f7:89  Channel: 17  NID: 0x23  TX-Power: 0dBm  State: IDLE
2015-05-10 14:10:51,695 - INFO #            Long HWaddr: 9a:17:6c:81:4d:4e:4c:51
2015-05-10 14:10:51,699 - INFO #            AUTOACK  6LO 
2015-05-10 14:10:51,703 - INFO #            Source address length: 2
2015-05-10 14:10:51,709 - INFO #            inet6 addr: ff02::1/128  scope: local [multicast]
2015-05-10 14:10:51,715 - INFO #            inet6 addr: fe80::ff:fe00:f789/64  scope: local
2015-05-10 14:10:51,716 - INFO #            
ping6 1 fe80::ff:fe00:e584
2015-05-10 14:11:20,396 - INFO # > ping6 1 fe80::ff:fe00:e584
2015-05-10 14:11:20,397 - INFO # 
2015-05-10 14:11:20,403 - INFO # icmpv6: Building ICMPv6 message with type=128, code=0
2015-05-10 14:11:20,407 - INFO # ipv6: NG_NETAPI_MSG_TYPE_SND received
2015-05-10 14:11:20,412 - INFO # ipv6: set payload length to zu (network byteorder 000c)
2015-05-10 14:11:20,415 - INFO # ipv6: set next header to 58
2015-05-10 14:11:20,419 - INFO # ipv6: calculate checksum for upper header.
2015-05-10 14:11:20,423 - INFO # ipv6: add to interface header to packet
2015-05-10 14:11:20,427 - INFO # ipv6: send unicast over interface 7
2015-05-10 14:11:20,430 - INFO # ipv6: send to 6LoWPAN instead
2015-05-10 14:11:20,433 - INFO # 6lo: NG_NETDEV_MSG_TYPE_SND received
2015-05-10 14:11:20,436 - INFO # 6lo: Send uncompressed
2015-05-10 14:11:20,440 - INFO # 6lo: max_frag_size = 127 for interface 7
2015-05-10 14:11:20,446 - INFO # 6lo: Send SND command for 0x200019ac to 7
2015-05-10 14:11:20,447 - INFO # 6lo: waiting for incoming message.
2015-05-10 14:11:20,455 - INFO # ipv6: waiting for incoming message.
2015-05-10 14:11:25,477 - INFO # ping timeout
2015-05-10 14:11:25,482 - INFO # --- fe80::ff:fe00:e584 ping statistics ---
2015-05-10 14:11:25,486 - INFO # 1 packets transmitted, 0 received, 100% packet loss
```

As stated above: up until to `ng_at86rf2xx`'s `_send()` function everything is in order. Here is the sniff of the packets above: https://www.cloudshark.org/captures/1cae4195d24f (source should be `fe80::ff:fe00:f789` but it is `4040:fe60::c3a:40fe:8000:0`).

As long as I am on the same binary the changed address will stay the same, no matter what. If I compile new it changes most of the time.

authmillenon, 2015-05-10:
I observed this on both `samr21-xpro` and `iot-lab_M3`

bapclenet, 2015-06-28:
@authmillenon, is this bug solved?

authmillenon, 2015-06-29:
Didn't test it in a while, but since we still have problems with wrongful overwriting in the packet buffer, I would dare to say it is still an issue.

gebart, 2015-06-29:
Does it happen with UDP as well as ICMPv6?

gebart, 2015-06-29:
@authmillenon Do you always get an `FE 80` at offset 11 in the source address?

gebart, 2015-09-23:
@authmillenon Is this still occurring in your tests?

authmillenon, 2015-09-23:
Not that I know of. Let's close this issue and reopen if someone sees this again

closed 2015-09-23 by authmillenon