RIOT-OS/RIOT #2965: nameless stack: possible duplicate entries

OlegHahm, 2015-05-11:
In several places in the new, nameless network stack arrays are used to manage lists of information such as IPv6 addresses for a given interface. Whenever an entry is removed from this list, it may happen that the list gets holey. However, the current implementation of the add functions usually only look for the first unused entry. This may lead to duplicate entries in the list.

Example:
List is: `A -> B -> C -> NULL -> NULL`
`remove(B)` =>  `A -> NULL -> C -> NULL -> NULL`
`add(C)` will find the first NULL entry (the second) and add `C` again:  `A -> C -> C -> NULL -> NULL`

One concrete occurrence is in [ng_ipv6_netif.c](https://github.com/RIOT-OS/RIOT/blob/master/sys/net/network_layer/ng_ipv6/netif/ng_ipv6_netif.c#L37)

cgundogan, 2015-05-11:
Makes sense.
What would be a good solution regarding this problem in your opinion? Simply iterating over all entries in the array to check for the appropriate ip address? And cache the first empty position encountered to use it in case it is not contained?

authmillenon, 2015-05-11:
This might be the best solution. Filling the holes on deletion, might lead to problems, since I keep the pointers to both the `ng_ipv6_nc_t` and `ng_ipv6_netif_t` entries in timeout timers in the neighbor discovery e.g. Moving the contents around might break stuff in these cases.

cgundogan, 2015-05-11:
>Filling the holes on deletion, might lead to problems..

+1 Yes, I am also against moving data around to fill gaps. I would claim that in the typical IoT scenario IP addresses wouldn't be added or deleted that often

Lotterleben, 2015-05-12:
Just wondering, would it make sense to keep a list of the instances where this occurs in this issue so we have a nice overview and also kind of a “to do” list?

authmillenon, 2015-05-12:
Added to the top-message

Lotterleben, 2015-05-21:
I think ``ifs`` in ``ng_netif.c`` is another occurence...

authmillenon, 2015-05-23:
This is not done yet ;)

Lotterleben, 2015-05-25:
Whoops, how did that happen? :o

authmillenon, 2015-05-25:
Auto-close, becauswle the PRs were somewhere marked as fixing ;) (via the words "fixes #<pr#>" in the comments e. g.)

authmillenon, 2015-05-25:
But afaik this issue is fixed now. So I close.

OlegHahm, 2015-05-25:
Great work, @Lotterleben and @authmillenon!

authmillenon, 2015-05-25:
My biggest contribution was the bug in the first place, so all the honor should go to @Lotterleben ;-)

OlegHahm, 2015-05-25:
Reviewing is also a valuable contribution.

Lotterleben, 2015-05-25:
waitwaitwait, what about ``ifs`` in ``ng_netif.c``? I'd reckon this is another instance of that error and would be happy to fix it, but stop me if I'm wrong here. ^^

authmillenon, 2015-05-25:
Oh yes, you're right.

authmillenon, 2015-05-25:
Added it to the top.

Lotterleben, 2015-05-25:
Okay, will take care of it as soon as possible!

OlegHahm, 2015-12-02:
I guess we won't manage to fix the missing parts for this release?

authmillenon, 2015-12-03:
Isn't this fixed by now?

OlegHahm, 2015-12-03:
I have no idea.

authmillenon, 2015-12-03:
Yapp it is.

closed 2015-12-03 by authmillenon

Lotterleben, 2015-12-03:
Pretty sure it is :)