RIOT-OS/RIOT #2967: Makefile.features

bapclenet, 2015-05-11:
In #2918, we've observed that the features written in board/XXX/Makefile.features is not always related to the board but mainly to the cpu. 
I think that two Makefile.features should exist. For instance, ``boards/samr21-xpro/Makefile.features`` and ``cpu/samd21/Makefile.features``

Most of those features (below) depends on the cpu and not the board.
``FEATURES_PROVIDED += transceiver``
``FEATURES_PROVIDED += periph_gpio``
``FEATURES_PROVIDED += periph_spi``
``FEATURES_PROVIDED += cpp``
``FEATURES_PROVIDED += periph_timer``
``FEATURES_PROVIDED += periph_uart``
``FEATURES_PROVIDED += periph_i2c``
``FEATURES_PROVIDED += periph_rtc``
``FEATURES_PROVIDED += periph_rtt``
``FEATURES_PROVIDED += periph_cpuid``
``FEATURES_MCU_GROUP = cortex_m0``

Then, ``FEATURES_CONFLICT`` will be added in one of the two files according to the reason of the conflict.

Kijewski, 2015-05-11:
[Makefile.buildtests:225](https://github.com/RIOT-OS/RIOT/blob/2682c38ee87f8f093693894757339a0a099adf76/Makefile.buildtests#L225) and [Makefile.include:287](https://github.com/RIOT-OS/RIOT/blob/2682c38ee87f8f093693894757339a0a099adf76/Makefile.include#L287) are the relevant lines that pull in `$(RIOTBOARD)/$(BOARD)/Makefile.features`. :+1: for adding `$(RIOTCPU)/$(CPU)/Makefile.features`.

bapclenet, 2015-05-12:
Ok great! :-)

Kijewski, 2015-05-12:
I found a problem with this solution: In Makefile.buildtests we don't know the CPU of a BOARD. So the BOARD's Makefile.features would need to CPU's Makefile.features. Does someone see a problem with that solution?

bapclenet, 2015-05-20:
@Kijewski, I didn't get your solution. Could you provide an example please?

OlegHahm, 2015-09-03:
@Kijewski, ping!

cgundogan, 2016-01-05:
@Kijewski ping ping

Kijewski, 2016-01-05:
`boards/msba2/Makefile.features` imports `boards/msba2-common/Makefile.features` imports `boards/msba2-common/Makefile.features` import `cpu/lpc2387/Makefile.features` import `cpu/arm7_common/Makefile.features`.

Rationale: `Makefile.buildtests` does not know MSB-A2's CPU.

kYc0o, 2016-08-04:
@Kijewski any news?

miri64, 2016-10-13:
Why was this marked as a known issue. Doesn't seem like a bug to me...

pyropeter, 2017-03-17:
While working on #6162 (Rust support) we stumbled upon this issue. The solution provided by @Kijewski  seems to be reasonable.

We tried to add a feature-flag for Rust support. Since the support depends on the cpu architecture, we added `FEATURES_PROVIDED += rust_support` to `cpu/native/Makefile.features` (and others). This works fine regarding the unsupported feature warning displayed when building code for an unsupported board. But this does not work as expected as far as the `info-boards-supported` target is concerned. As @Kijewski explained, this is caused by the `board_missing_features` function in `Makefile.buildtests`. It does not include `cpu/$cpu/Makefile.features`. This is not easy to fix because the CPU of a certain board is not known in that function.

pyropeter, 2017-03-23:
Hello?

kaspar030, 2017-03-23:
> Doesn't seem like a bug to me...

This is a bug, because Makefile.features is currently not considered for "make info-boards-supported".
This doesn't manifest, as all Makefile.features within boards only define periph_pm, which is currently not correctly used anywhere.

kaspar030, 2017-11-02:
Fixed by #7880.

closed 2017-11-02 by kaspar030