RIOT-OS/RIOT #2971: Module `ng_icmpv6` and `ng_icmpv6_echo`

cgundogan, 2015-05-12:
What is the purpose of defining these two modules in `Makefile.dep`? They both include `ng_ipv6`, which again includes `ng_icmpv6`. Smells like a circular dependency.. couldn't we just drop them and use `ng_ipv6` instead?

authmillenon, 2015-05-12:
`ng_ipv6` and `ng_icmpv6` have a circular dependency. ICMPv6 without IPv6 does not make any sense and IPv6 needs ICMPv6 to function properly (though it is totally fine without it too). But I fail to see why this is a problem. `ng_icmpv6_echo` is only dependent on `ng_icmpv6` not vice versa so there is no circular dependency.

cgundogan, 2015-05-12:
Right. Am I wrong assuming then that `ng_ipv6` and `ng_icmpv6` are module-wise completely equivalent, because they both have the same (transitive) dependencies?
Why would anyone use `ng_icmpv6` in their Makefile, when `ng_ipv6` also includes the same deps?

authmillenon, 2015-05-12:
I don't know. Some users do weird stuff ;)

cgundogan, 2015-05-12:
Therefore, I started this discussion... to prohibit weird stuff by not allowing it in the first place? (:

authmillenon, 2015-05-12:
How do you want to disallow (mechanically, not conventionally, because rules by convention can be broken easily ;)) this, so a new user would know it is not allowed?

kaspar030, 2015-05-12:
Why not keep the dependencies as they are (icmpv6 requires ipv6 but not the other way around), and create a pseudo-module for the whole stack, which in turn depends on what 99% of the users will want?

That way users could either USEMODULE += ipv6_full, or hand-pick their components. Keeps most of our Makefiles shorter, too.

authmillenon, 2015-05-12:
Sounds good. Will open a PR tomorrow.

kaspar030, 2015-05-12:
or maybe ipv6_default isntead of ipv6_full. Same for sixlowpan + sixlowpan_default (the first depending on ipv6, the latter depending on ipv6_default)?

OlegHahm, 2015-05-12:
That could also solve #2947.

PR #3004 (2015-05-17): make: introduce *_default pseudo-modules
Fixes #2971

Follow-up to #2910 since I think NDP belongs in here, too.
commit d76fb3fa (2015-05-17): make: introduce *_default pseudo-modules

Fixes #2971
closed 2015-05-17 by authmillenon