RIOT-OS/RIOT #2972: Cortex-M4: Stack alignment

gebart, 2015-05-12:
I'm hitting a bug where Clang assumes that the stack upon function entry is 8 byte aligned. GCC does not trigger the broken behaviour.

The following code exists in auto_init.c:
```c
        memcpy(&(eui64.uint32[0]), &hash_h, sizeof(uint32_t));
        memcpy(&(eui64.uint32[1]), &hash_l, sizeof(uint32_t));
```

Clang compiles this to:

```asm
  3e:	a901      	add	r1, sp, #4
  40:	2204      	movs	r2, #4
  42:	4628      	mov	r0, r5
  44:	f7ff fffe 	bl	0 <memcpy>
  48:	f045 0004 	orr.w	r0, r5, #4 <<<<<------ OR instead of ADD 
  4c:	4669      	mov	r1, sp
  4e:	2204      	movs	r2, #4
  50:	f7ff fffe 	bl	0 <memcpy>
```

which makes the second memcpy actually copy to `eui64.uint32[0]` instead of `eui64.uint32[1]`

All stack manipulation inside the function is aligned in a way that eui64 is aligned to an 8 byte offset from `$sp`, however, the stack pointer register is aligned to a 4-byte boundary upon entry, causing the OR operation before the second memcpy to have no effect.

GCC does not optimize the address computation into an OR, but rather use an ADD instead.
Clang generates an ADD if I define a uint32_t pointer variable and assign it the value of `&eui64.uint32[0]` and then use `var+1` as the destination of the second memcpy, but that's just a workaround for the real problem.

I am investigating right now if the process stack definition or the linker script can be changed to make sure the stack pointer is aligned on an 8 byte boundary.

The Procedure Call Standard for the ARM Architecture (AAPCS) states that the stack is aligned on a 8 byte boundary at function entry and exit.

References:
AAPCS advisory: http://infocenter.arm.com/help/topic/com.arm.doc.ihi0046b/IHI0046B_ABI_Advisory_1.pdf

gebart, 2015-05-12:
After some more digging it seems like after the return from isr_svc the stack pointer is always aligned on a non-8 byte aligned 4 byte alignment (4 or c as the last digit of the address)

gebart, 2015-05-13:
Found the bug in the initial stacked xPSR when creating a thread, PR #2976 fixes it.

OlegHahm, 2015-05-18:
#2976 is merged

closed 2015-05-18 by OlegHahm