RIOT-OS/RIOT #2988: ng_inet_csum: incorrect checksum calculation

authmillenon, 2015-05-15:
There are reproducable instances, where the the checksum of ICMPv6 (and probably UDP - though not confirmed) packets is wrong: This includes some echo requests ([see the two last ones here](https://www.cloudshark.org/captures/d0aeb1504620), interestingly enough the 4th has the checksum has the checksum the 3rd was expecting) and echo replies, when testing #2910 with two nodes on `ng_nativenet`: https://www.cloudshark.org/captures/b985f35e13a1 (packet 4).

@sgso also observed echo requests where the checksum will always be off by `0x700`.

OlegHahm, 2015-05-15:
I'm not sure how I can help...

authmillenon, 2015-05-15:
TBH I assigned you mainly because I did not know any other one. I might look into it after MobiSys, but I won't be able before next week.

authmillenon, 2015-05-15:
Just to give a hint: the checksum is definitly calculated incorrectly, it's not one of these "a value is different then before" issues. I checked this.

OlegHahm, 2015-05-15:
>  it's not one of these "a value is different then before" issues. I checked this.

What do you mean?

authmillenon, 2015-05-15:
The value `ng_icmpv6_calc_csum()` puts into the header is the same, that Wireshark reports as wrong.

authmillenon, 2015-05-16:
#2998 should help

PR #2998 (2015-05-16): ng_icmpv6: pre-set checksum field to 0 for sending
Hopefully fixes #2988.
closed 2015-05-16 by authmillenon