RIOT-OS/RIOT #3007: ng_sixlowpan: release sent fragment snippets

OlegHahm, 2015-05-16:
As discussed in https://github.com/RIOT-OS/RIOT/pull/2781#discussion_r29076938 already sent fragment snippets should be released from the packet buffer immediately after sending. This requires to keep somehow track of the current offset.

authmillenon, 2015-05-16:
At least with the current `pktbuf` implementation (and any implementation utilizing POSIX-conforming `free()` for a dynamically allocated packet buffer) this would not give any benefit, since nothing is actually released from the packet buffer, until the whole chunk (yes, chunk, not snip*) the data is in is actually released.

* snips are protocol-logical units in a packet, chunks are memory-logical units in the packet buffer (I made this definition up right now ;-)). A chunk is allocated on packet creation in the packet buffer. If a snip gets split up, the chunks stays the same, the packet buffer just notes that the overlaying snip got split. This is necessary since e.g. `free()` does not support partial releasing of the memory, so information on the originally allocated data chunk needs to be stored somewhere.

authmillenon, 2015-05-16:
However for #3008 this behavior could still be beneficial ;-)

OlegHahm, 2015-05-16:
I don't understand. Does not `pkt` represent a "chunk"?

authmillenon, 2015-05-16:
First of all, there is no datatype `pkt`, it is just a list of `pktsnip_t`. But regarding `pktsnip_t`: Not necessarily: a chunk is any new instance of `pktsnip_t::data` to the packet buffer (basically any time you call `ng_pktsnip_add()` and the `data` argument is not already in the packet buffer). If `pktsnip_t::data` however is already known to the packet buffer (e.g. a protocol threads marks its header in a received packet), there is no new chunk created, but it is "split" (at least that's what the user perceives). Internally, this is currently handled by counting the packet snips referencing this certain data chunk. On every release (that reaches 0) to a packet snip, that references this chunk this counter is decreased. Only if the last referencing snip is released completely the chunk gets removed from memory.

authmillenon, 2015-05-16:
> (e.g. a protocol threads marks its header in a received packet)

The mechanism marking the already send parts would look similar ;-)

OlegHahm, 2015-05-16:
> First of all, there is no datatype pkt, it is just a list of pktsnip_t.

I was referring to the particular piece of code, cited in my first post.

OlegHahm, 2015-05-16:
I still don't understand how you explanation is related. I have several pieces of memory representing the fragments. Whenever one is sent it should be freed.

authmillenon, 2015-05-16:
That's what I am trying to explain: quoting from https://github.com/RIOT-OS/RIOT/issues/3007#issuecomment-102704709 "with the current pktbuf implementation (and any implementation utilizing POSIX-conforming free() for a dynamically allocated packet buffer)" this is not the case. You can't just free memory arbitrarily, you have to free it referring to the pointer you allocated it with (which in this case is the start of the IPv6 packet or its payload).

OlegHahm, 2015-05-16:
But the fragments are separate memory pieces, that has been allocated by the packet buffer before. In my understanding of the packet buffer and malloc-free I can free any piece of memory that I have reserved beforehand.

authmillenon, 2015-05-17:
What are memory pieces in this context?

The state of the packet buffer before fragmentation is basically: 

* 20-byte pktsnip_t, 
* x Byte payload, 
* 20-byte pktsnip_t, 
* x Byte L4 header, 
* 20 Byte pktsnip_t, 
* 40 Byte IPv6 header, 
* 20 Byte pktsnip_t, 
* x Byte L2 header
* (plus some additional internal Metadata and even other packet snips in-between).

Each item in that list (except the last one of course) is a chunk allocated by the upper layers. If we fragment this, the fragment boundaries will very likely be in the "x Byte payload" chunk (which in that case will also be the significantly largest). We have to retain this chunk in the malloc/free-pktbuf until all of its fragments are send, because `free()` works only on the exact pointer, that `malloc()` (and friends) returned. You even can't call free multiple times on the same segment. The first call frees it; always. It is simply not possible to just free parts of a malloc'd memory chunk. I kept this paradigm in the static packet buffer also to be compatible to the dynamic one (and because it was simpler to implement as a proof-of-concept at the moment of implementation).

OlegHahm, 2015-05-17:
> the fragment boundaries will very likely be in the "x Byte payload" chunk (which in that case will also be the significantly largest

That's what I'm talking about: i read the code in l90 as if fragments are allocated in the packet buffer as isolated snippets.

authmillenon, 2015-05-17:
No, that are the headers of the packet.

authmillenon, 2015-05-17:
I'm fragmenting the packet in that function. Where should it be fragmented previously?

OlegHahm, 2015-05-17:
Why previously?

authmillenon, 2015-05-17:
Ah, you meant they are allocated there. But those are the parts for sending… these ARE released. By the driver, as any other snip going to the driver. So what is this issue about? I thought it was about releasing parts of the still assembled IPv6 packet.

OlegHahm, 2015-05-17:
Aah, then we can probably close this.

PeterKietzmann, 2017-01-31:
I close this for @OlegHahm.

closed 2017-01-31 by PeterKietzmann