RIOT-OS/RIOT #3009: ng_sixlowpan: reserve full packet only after reception of all fragments

OlegHahm, 2015-05-16:
Currently the implementation of 6lowpan fragmentation reserves space for the full packet after reception of the first fragment. I would suggest to do so only at the end, after all fragments have arrived. This would make it possible to handle fragments from different packets concurrently.

authmillenon, 2015-05-16:
This feature would require a comparable study of the throughput in such a scenario (multiple packets arriving concurrently but with fragments out-of-order), too. Re-sizing the data segment of a packet snip can lead to movement of the data in the packet buffer if it won't fit in the previous space, if fragments of multiple packets are arriving concurrently this might produce more time-overhead due to copying the data around in the packet buffer on (worst-case) every fragment receive compared to just waiting for a potential resent.

A more `pktbuf`-ique alternative could be to store every received fragment payload as a new pktsnip, but this could lead to potentially more memory overhead in the packet buffer. Keep in mind, that every `ng_pktsnip_t` struct costs an extra 20 byte in the packet buffer.

authmillenon, 2015-05-16:
I'm also not sure right now, how upper-layers would react to the second alternative.

authmillenon, 2015-05-16:
(probably not good, since most of them basically assume that the received payload is one coherent snip, it also would produce problems if an upper-layer header is divided between multiple fragments)

OlegHahm, 2015-05-16:
I think it's also a security concern. Currently, a node can pretty easily be DOS'ed by just sending peridocially fragments with with differing tags and a big packet length field.

authmillenon, 2015-05-16:
> I think it's also a security concern. Currently, a node can pretty easily be DOS'ed by just sending peridocially fragments with with differing tags and a big packet length field.

That's only partly true. The reassembly buffer itself is very least effort by design and specification. If no space is left in it, it just kicks out the oldest entry (deleting all potentially allocated packets). Currently this is not the case for the "packet buffer is full" case on reassembly buffer allocation, however. We could however optimize the reassembly buffer a little bit in that direction to minimize this security concern ;)

miri64, 2019-08-10:
This could still be interesting to implement. Having different snips isn't that much of an issue anymore, taking `gnrc_pktbuf_merge()` into account (this would duplicate only once when the datagram is complete; we duplicate right now as well hpwever, just per fragment)

miri64, 2021-03-20:
We still can look into this, however in the first version of the VREP, presented [in this paper](https://doi.org/10.1145/3405656.3418719), I did exactly that and the packet buffer usage was massive compared to the final version or just hop-wise reassembly.

closed 2022-04-19 by stale[bot]

miri64, 2022-04-19:
Should have been already closed after https://github.com/RIOT-OS/RIOT/issues/3009#issuecomment-803298568