RIOT-OS/RIOT #3020: board: RIOT port for La BlueFrog

PeterKietzmann, 2015-05-18:
List to coordinate porting efforts for the [La BlueFrog](http://www.la-bluefrog.io/) board. The cpu (stm32l1) of the current revision is already supported in RIOT. Please leave a comment what you are willing to work on. Also feel free to adapt and extend the list below.

Porting ToDos:
* [ ] basic file structure (flashing, debugging, CPU-header, makefile, peripheral configuration, ..)
  - [ ] testing of peripheral drivers (GPIO, I2C, SPI, UART, Timer/PWM, Analog)
* [ ] LED macros
* [ ] sensor drivers
  - [ ] ST VL6180X ambient light sensor, proximity/distance densor
  - [ ] ST LSM6DS3 3-axis accelerometer, 3-axis gyroscope (compare [lsm303dlhc](https://github.com/RIOT-OS/RIOT/tree/master/drivers/lsm303dlhc))
  - [ ] ST LIS3MDL 3-axis magnometer (compare [lis3dh](https://github.com/RIOT-OS/RIOT/tree/master/drivers/lis3dh))
  - [ ] ST SLPS25H atmospheric pressure (and altitude) sensor
  - [ ] Knowles SPU0414HR5H-SB microphone (as ambient sound sensor)
* [ ] OLED display driver
* [ ] PAN1740 BLE driver
* [ ] BLE adaption ?
* [ ] short wiki documentation

PeterKietzmann, 2015-05-18:
@katezilla what du you think about working on the first three main points?

PeterKietzmann, 2015-05-18:
@YRaoul here is my suggested ToDo list to add support for La BlueFrog in RIOT. Feel free to edit, expand oder assign yourself to a task.

kaspar030, 2015-05-18:
Consider working directly with #2852. It should be merged soon. :)

PeterKietzmann, 2015-05-18:
Good point. Thanks fot the reference!

katezilla, 2015-05-19:
First two points sound interesting, i can do those. Third point might be too much, i would do one or two sensors.

PeterKietzmann, 2015-05-19:
Well, I think that's fine! Feel free to edit the list above and assign yourself to a task (by writing your nick behind the description). When you open a PR you should link it there too. But at first let's wait for the board to arrive :-)

YRaoul, 2015-05-20:
Hello, thanks for the ToDo list.
I could take  "OLED display driver", but we maybe need to have some discuss on this, as I don't know if RIOT has already a "display driver" feature.
RIOT could provide a printf service on the display, but I don't know if it will be fast enough to be usefull.
And maybe we need to provide some "screen sharing" feature between OS and App code...
Hope your board will arrive pretty soon !

haukepetersen, 2015-05-20:
Hi. Regarding the display driver, we do not really have something in RIOT. I did a driver for a very simple b&w display some month ago (-> see `drivers/pcd8544`). But I did not separate any generic display API or similar. For porting the BlueFrog display, I would suggest to start with a specific driver just for the specific display and extract a more generalized API from this in a second step.

PeterKietzmann, 2015-05-21:
Yes, that seems valid to me. I'd suggest to just start writing the driver (when there is a rudimentary RIOT port)  in a very modular way. Some time ago I already thought a display driver interface for RIOT would be cool. Anyway, if you have any questions about developing for RIOT, just post here, in the PR you'll open for your driver or on the mailing list.

ReneHerthel, 2015-06-16:
i would like to implement the LSM6DS3 sensor.

PeterKietzmann, 2015-06-16:
Nice! Just edit the above list by clicking on the small pencil in the upper right corner and place your github name to the related point. Feel free to open a PR any time and link it there too.

katezilla, 2015-06-16:
@PeterKietzmann I am not so sure i can edit your post. Can you add me to the first two points for now?

PeterKietzmann, 2015-06-16:
Did. Wasn't aware that you can't edit it.

ReneHerthel, 2015-07-21:
@PeterKietzmann I would like to prepare the LIS3MDL driver. This one looks much easier to implement than the LSM6DS3, so it's a good idea when I implement both.

PeterKietzmann, 2015-07-21:
Ok. I put your name in the list above. Don't forget to check the similarities to the LIS3DH. (If they are really similar you could try to merge both drivers in one implementation. Otherwise just write an own)

PeterKietzmann, 2015-07-28:
Just for the record: Former "La BlueFrog" is "LimiFrog" now. The ongoing port is based on hardware revision 1, therefore we'll name it "limifrog-v1" or similar. The current hardware revision has an other MCU and will be ported any time in future. I adapted the titles in the regarding PRs (even if the drivers should be independent from that board, it is the only board available for us, providing these sensors)

PeterKietzmann, 2015-07-30:
@YRaoul the initial PR for the LimiFrog port in RIOT has been open #3509. Do you still plan to spend work on a display driver?

katezilla, 2015-08-25:
Added limifrog-v1 to the wiki, see [here](https://github.com/RIOT-OS/RIOT/wiki/Board:-LimiFrog-v1).
I will keep on working on the BLE driver :)

ReneHerthel, 2015-11-09:
@PeterKietzmann I work for the SLPS25H driver at the moment

ReneHerthel, 2016-01-06:
@PeterKietzmann I would like to complete this and write the other drivers for this port. Did you have an idea who can help me testing this now? 
And what about the BLE driver?

PeterKietzmann, 2016-01-08:
synchronized offline

PeterKietzmann, 2017-11-06:
This is not WIP anymore. Basic support has been provided. Open checkboxes won't be implemented as part of this project so I'll close the issue now.

closed 2017-11-06 by PeterKietzmann