RIOT-OS/RIOT #3039: 802.15.4(e) MAC-Layer Task Force

jremmert-phytec-iot, 2015-05-20:
There is the consensus that we need some kind of duty cycling MAC layer within RIOT. Within this issue we can provide a collection of open topics and progression. Feel free to add open points, topics and PRs.

## Overview and open PRs:
- [x] nomac [merged -> #2426]
- [ ] csma_mac (@jremmert-phytec-iot) [pr -> #2467] - *(currently outdated and based on depricated assumptions, will update soon)*
- [ ] lpm [pr -> #3019, #2309, #2650]

## Current Issues:
### Message dispatching
When a node will be associated to an PAN-coordinator, the transmitting and receiving time is guided on the superframe architecture. That means there will be timespans, where the node can not send messages. Nevertheless, there will be incoming task-messages from upper-layer to the MAC layer task.
1. Messages from upper layer              (Low-prio)
2. Messages from driver (isr-events)    (High-prio)
3. Messages from timer                        (High-prio)
There must be a mechanism that (i) keeps the message queue empty. (ii) Takes care of the current MAC-layer state (just try to send a new packet when the MAC layer handled the previous one).

### Storing state variables for the MAC-layer state machine.
1. Static variables: Problem when we have more than one MAC-layer
2. Store them in the device-descriptor: The same device-descriptor may be used in different MAC-layer. How to choose the fitting variable set (at compile time)?
3. ? 

## Open Topics:
1. Move the packet assemble from IEEE 802.15.4 compatible device-drivers to the MAC-layer.
1. Procedure to configure the MAC-layer and start device discovery and associate with PAN (some Auto-init procedure). 
 * Initialize *capability information field* according to the device capability [IEEE 802.15.4 , section 5.3.1.2].
 * Notify upper layer about established link (associate successfull).
2. Implement MAC commands (MLME management service) [IEEE 802.15.4 standard section 6.2]
 * Scan
 * Associate, Disassociated
 * ...
2. Implement TSCH MAC commands (MLME management service) [IEEE 802.15.4e standard section 6.2.0b]
 * ...

OlegHahm, 2015-05-20:
I will probably work on extracting the MAC part from OpenWSN soonish.

bapclenet, 2015-05-20:
What about hardware CSMA? AT86rf2xx provides CSMA (when it will be implemented). 
Will it be possible to choose between hardware CSMA and software CSMA?

OlegHahm, 2015-05-20:
> Will it be possible to choose between hardware CSMA and software CSMA?

Once we have a software implementation for CSMA, that should be pretty simple.

jremmert-phytec-iot, 2015-05-20:
> I will probably work on extracting the MAC part from OpenWSN soonish (which is 802.15.4e).

Cool :+1: , that would be very interesting to us!

> What about hardware CSMA? AT86rf2xx provides CSMA (when it will be implemented). 
Will it be possible to choose between hardware CSMA and software CSMA?

The current AT86rf2xx already implements these HW CSMA functions (CCA, Backoff-wait on channel access failure, acknowledge handling and retransmission).

My current approach would to decide accordingly to the driver responses what mechanism to use (HW or SW-CSMA). If all necessary options for HW-CSMA are available in the radio driver and could be set successfull, most of the CSMA-MAC layer mechanisms will be omitted (e.g. in the Atmel transceiver). E.g. for the Freescale Kinetis KW2x´ internal transceiver we need to choose some more SW-CSMA mechanisms.

daniel-k, 2015-05-21:
Does (ng_)netdev already have some API for CSMA and CCA? Would it make sense to implement a second API next to netdev that implements radio specific functions like CSMA? Having a "sublayer" like this would make it easier to implement SW+HW CSMA in a more generic way I guess.

jremmert-phytec-iot, 2015-05-21:
In far as I know, ng_netdev is ment as an universal interface between a MAC-layer and driver. I don´t see that a second api next to ng_netdev would make SW+HW CSMA easier. 

What I  (rudimentary) did is asking the driver via ng_netdev option get/set if it is capable of HW-CSMA/ACK/Resend Mechanisms. If yes, the MAC-layer will behave like the current ng_nomac layer. If not, a statemachine will be used that controls the steps neccessary for SW-CSMA/ACK/Resend. So, it won´t make a huge difference if an advanced MAC-layer will be based on the ng_nomac-layer or an ng_csma-MAC layer.

To get a picture of the idea you could have a look at my current working branch. I will always be thankful for hints on that: https://github.com/jremmert-phytec-iot/RIOT/tree/wip%40csma_mac

daniel-k, 2015-05-28:
So we agree that there needs to be duty cycling to save energy. But that's a topic concerning the whole system, not just the MAC layer or just the network stack. I see there's an API for power management, at least to set power modes, but is there also a concept on how to decide when to go to low power modes?

I wonder who (component-wise speaking) is in charge of putting the MCU to sleep. After some thoughts about it I came to the conclusion, that no module can have this power alone, because it will always have side-effects for the others. There should be some kind "request-awake" and "release-awake", so that the idle thread may put the controller to sleep when there are no more awake requests.

What do you think? Is there maybe already something similar? This seems quite generic, so this should maybe discussed somewhere else? I posted here because I thought about it in the context of my MAC layer implementation.

daniel-k, 2015-05-28:
I just found https://github.com/RIOT-OS/RIOT/issues/2927 which seems to kind of match my proposal

OlegHahm, 2015-05-28:
> I wonder who (component-wise speaking) is in charge of putting the MCU to sleep.

The IDLE thread.

daniel-k, 2015-10-19:
Concerning hardware CSMA I feel that [this discussion](https://lists.riot-os.org/pipermail/devel/2015-October/003069.html) on the mailing list should be linked here.

PeterKietzmann, 2017-11-27:
@jremmert-phytec-iot, @daniel-k, @zhuoshuguo could anyone of you give an estimation about this issue? What does the current situation look like and how plans to spend further time in what?

zhuoshuguo, 2017-11-29:
For duty-cycling MAC protocol implementations, currently, we have LWMAC in RIOT, and now [GoMacH](https://github.com/RIOT-OS/RIOT/pull/5618) is close to be merged.

PeterKietzmann, 2018-06-26:
All referenced PRs have been merged or closed. Closing this issue now. Please re-open if I'm mistaken.

closed 2018-06-26 by PeterKietzmann