RIOT-OS/RIOT #3040: Replacing /n to /r/n in output statements

rakendrathapa, 2015-05-21:
printf and puts statement outputting to the terminal should use /r/n instead of just /n for new line. Some of the terminals do not put carriage return by default and to make it compatible across all the serial consoles. Else some test cases prints just fall off the boundary.

closed 2015-05-21 by rakendrathapa

gebart, 2015-05-21:
What terminal program are you using? Most serial programs that I have used have a setting for treating \n as crlf. Many places in the code use puts() to output text, which automatically adds newline, but no carriage return, to the output stream. It would be very unintuitive if we were to change all strings to end in \r.

rakendrathapa, 2015-05-21:
Well, for embedded boards, I have always adopted the policy to consider serial terminals as dumb and always include carriage return to remove dependency on them. 
But yes, considering so many printf / puts statements, it is counter intuitive to include carriage return in every statement rather than just configuring serial terminal. That is why I closed the issue. :+1: