RIOT-OS/RIOT #3052: NDP: implement missing functionality

authmillenon, 2015-05-23:
This is an overview issue to organize ongoing development of IPv6 neighbor discovery:

- [ ] Main ND, [RFC 4861](https://tools.ietf.org/html/rfc4861)
  - [ ] Router Discovery and Prefix Discovery; [section 6](https://tools.ietf.org/html/rfc4861#section-6) [#3049 -> under review] 
  - [x] Address Resolution and Neighbor Unreachability Detection; [section 7](https://tools.ietf.org/html/rfc4861#section-7) [#2910 -> merged]
  - [ ] Redirect Function; [section 8](https://tools.ietf.org/html/rfc4861#section-8)
- [ ] IPv6 Stateless Address Autoconfiguration, [RFC 4862](https://tools.ietf.org/html/rfc4861)
- [ ] Neighbor Discovery Optimization for 6LoWPANs, [RFC 6775](https://tools.ietf.org/html/rfc6775) [WIP by @authmillenon]

miri64, 2016-10-17:
Should look into this when working on #5293.

PeterKietzmann, 2017-11-27:
@miri65 now that the new NDP is nearly there, could you update the "missing functionalities" list please

miri64, 2017-11-27:
> @miri65

I'm getting older :sob: 

This is still up-to-date. The NIB did not add any significant new functionality. It's in the project as a reminder to the testers, that some functionality isn't implemented yet (and won't be in #7925).

(side-note: even if there would be new functionalities, I only would update them *after* #7925 got merged ;-))

tinstructor, 2020-11-20:
@miri64 @benpicco I was wondering what happens when the ABRO valid lifetime expires since [RFC 6775](https://tools.ietf.org/html/rfc6775) is rather ambiguous (at least to me it is) on this subject. For example, it states on [page 29](https://tools.ietf.org/html/rfc6775#page-29): 

> When the ABRO Valid Lifetime associated with a 6LBR times out, all information related to that 6LBR MUST be removed.
 
But then on [page 36](https://tools.ietf.org/html/rfc6775#page-36) it says:

> The router keeps state for each 6LBR that it sees with an ABRO. This includes the version number, the Valid Lifetime, and the complete set of PIOs and 6COs. The prefixes are timed out based on the Valid Lifetime in the PIO. The Context Prefix is timed out based on the Valid Lifetime in the 6CO.

I've tried digging through the code and I think I understood that RIOT just times out all information stored for a given 6LBR (i.e., including contexts and prefixes) when the ABRO valid lifetime expires, regardless of the fact that prefix and context lifetimes may not have expired yet. Conversely I understood from the code that when prefixes and / or contexts expire before the ABRO does, they are timed-out independently. Is my assumption correct?