RIOT-OS/RIOT #3063: Discussion on how and where to store source routes

cgundogan, 2015-05-26:
@BytesGalore I opened this issue to not hijack PR #3058 for this discussion.

RPL's non-storing mode requires a data structure to store source routes at the root node.
During the second NSTF meeting there was a very brief discussion on where such a data structure should reside.

I can identify 2 different locations:
1. It should be implemented as a standalone entity within its own thread - global scope and accessable by other modules
2.  It should be implemented within the current `fib` implementation - the data structure itself would be inaccessable, but the `fib` would offer an API to modify the source routing table

I think @emmanuelsearch proposed at the NSTF meeting that such an implementation should be a standalone entity and globally accesable.

@BytesGalore may(?) be inclined to put this into the `fib` (#3058).

After sorting out where to put such a data structure, the question is how it should be implemented. Is it a linked-list, a tree structure, or something completely else?

To get some insights I looked into other RPL implementations, but I could not find any implementation, which was supporting RPL's non-storing mode. I also briefly checked linux' network stack implementation how source routes could be stored, but was not able to find it. (maybe related to: [RFC5095](https://tools.ietf.org/html/rfc5095)?)

OlegHahm, 2015-12-04:
Seems like a pretty short and one-sided discussion so far. @emmanuelsearch, @cgundogan, how are we going forward with this?

BytesGalore, 2015-12-04:
@OlegHahm basically the fib now stores and handles source route entries, so we have 2. available.

cgundogan, 2016-03-29:
Let's postpone this to the next release, because we still have no experience with the implemented source rout version of the fib. For the next release we hopefully get the non-storing mode of RPL that utilizes that fib feature, then we can discuss more.

kYc0o, 2016-07-26:
@cgundogan AFAIK there is no non-storing mode currently on master, so I'll postpone it for the next release.

closed 2020-03-13 by stale[bot]