RIOT-OS/RIOT #3069: shell: strange warnings for netif

OlegHahm, 2015-05-26:
I'm not exactly sure, what's going wrong here, but if I merge #2944 on current master and add
```diff
+USEMODULE += ng_sixlowpan_default
+USEMODULE += ng_ipv6_netif
```
to the `examples/default` Makefile, I receive these warnings:
```
"make" -C /home/oleg/git/RIOT/sys/shell/commands
/home/oleg/git/RIOT/sys/shell/commands/sc_netif.c: In function '_netif_send'
/home/oleg/git/RIOT/sys/shell/commands/sc_netif.c:59:16: warning: array subscript is above array bounds [-Warray-bounds]
         if (ifs[i] == dev) {
                ^
/home/oleg/git/RIOT/sys/shell/commands/sc_netif.c: In function '_netif_config'
/home/oleg/git/RIOT/sys/shell/commands/sc_netif.c:707:13: warning: array subscript is above array bounds [-Warray-bounds]
             _netif_list(ifs[i]);
             ^
/home/oleg/git/RIOT/sys/shell/commands/sc_netif.c:59:16: warning: array subscript is above array bounds [-Warray-bounds]
         if (ifs[i] == dev) {
                ^
/home/oleg/git/RIOT/sys/shell/commands/sc_ipv6_nc.c: In function '_ipv6_nc_manage'
/home/oleg/git/RIOT/sys/shell/commands/sc_ipv6_nc.c:94:16: warning: array subscript is above array bounds [-Warray-bounds]
         if (ifs[i] == iface) {
                ^
/home/oleg/git/RIOT/sys/shell/commands/sc_ipv6_nc.c: In function '_ipv6_nc_routers'
/home/oleg/git/RIOT/sys/shell/commands/sc_ipv6_nc.c:94:16: warning: array subscript is above array bounds [-Warray-bounds]
         if (ifs[i] == iface) {
 ```

Which is strange because `NG_NETIF_NUMOF` is properly set to 1. If I manually change the array size to 1, the warning remains, but vanishes if I set it to 2...

Is this some weird behavior of my toolchain or do I miss something?

authmillenon, 2015-05-26:
As discussed with you on Jabber, I suspect the return value of `ng_netif_get()` to be too big. Have you investigated this?

OlegHahm, 2015-05-26:
But how should the compiler be able to check at compile time?

authmillenon, 2015-05-26:
Oh right... This is a compiler issue.

OlegHahm, 2015-05-26:
Can someone check if this Bug occurs with other GCC versions? I'm using:
`arm-none-eabi-gcc (Arch Repository) 5.1.0`

OlegHahm, 2015-05-26:
Apparently, I don't get the warning with `arm-none-eabi-gcc (GNU Tools for ARM Embedded Processors) 4.8.4 20140526 (release) [ARM/embedded-4_8-branch revision 211358]`.

OlegHahm, 2015-05-27:
Wrong toolchain.

closed 2015-05-27 by OlegHahm