RIOT-OS/RIOT #3080: NSTF: Ping of death

OlegHahm, 2015-05-27:
On IoT-LAB this happens when ping two neighboring nodes after 1 to n times, where the biggest n that I observed was 7:
```
m3-100;ping6 fe80::ff:fe00:9616
1432715099.557171;m3-100;> ping6 fe80::ff:fe00:9616
1432715099.567645;m3-100;12 bytes from fe80::ff:fe00:9616: id=88 seq=1 hop limit=64 time = 0.9606 ms
1432715099.577621;m3-100;12 bytes from fe80::ff:fe00:9616: id=88 seq=2 hop limit=64 time = 0.7719 ms
1432715099.592549;m3-100;12 bytes from fe80::ff:fe00:9616: id=88 seq=3 hop limit=64 time = 0.13201 ms
1432715099.593531;m3-100;--- fe80::ff:fe00:9616 ping statistics ---
1432715099.594521;m3-100;3 packets transmitted, 3 received, 0% packet loss, time 0.0330526 s
1432715099.595440;m3-100;rtt min/avg/max = 0.7719/0.10175/0.13201 ms
m3-100;ping6 fe80::ff:fe00:9616
1432715100.917122;m3-100;> ping6 fe80::ff:fe00:9616
1432715100.952338;m3-120;> ******** SYSTEM FAILURE ********
1432715100.952675;m3-120;
1432715100.953008;m3-120;HARD FAULT
1432715100.953879;m3-120;	pid | name                 | state    Q | pri | stack ( used) | location   
1432715100.956623;m3-120;	  1 | idle                 | pending  Q |  15 |   256 (  136) | 0x200001dc 
1432715100.958571;m3-120;	  2 | main                 | bl reply _ |   7 |  2048 (  416) | 0x200002dc 
1432715100.959421;m3-120;	  3 | uart0                | bl rx    _ |   6 |  1024 (  196) | 0x20000bc0 
1432715100.961225;m3-120;	  4 | pktdump              | bl rx    _ |   6 |  2048 (  244) | 0x200031f8 
1432715100.963311;m3-120;	  5 | 6lo                  | bl rx    _ |   3 |  1024 (  368) | 0x200039fc 
1432715100.964291;m3-120;	  6 | ipv6                 | running  Q |   4 |  1024 (  528) | 0x200013ec 
1432715100.967098;m3-120;	  7 | udp                  | bl rx    _ |   5 |  1024 (  264) | 0x20003fd4 
1432715100.967799;m3-120;	  8 | at86rfxx             | bl rx    _ |   4 |  1024 (  356) | 0x20000fc0 
1432715100.968742;m3-120;	    | SUM                  |            |     |  9472 ( 2508)
1432715100.968993;m3-120;
1432715100.969766;m3-120;******** RIOT HALTS HERE ********
```

I ran a couple of L2 experiments last week on IoT-LAB which were super stable, so I would assume that is not a driver or hardware issue.

How to reproduce:
current master + #2895 + #3078 +
```diff
diff --git a/examples/ng_networking/Makefile b/examples/ng_networking/Makefile
index 161351c..5a3f110 100644
--- a/examples/ng_networking/Makefile
+++ b/examples/ng_networking/Makefile
@@ -8,10 +8,13 @@ BOARD ?= native
 RIOTBASE ?= $(CURDIR)/../..
 
 # Specify the modules that are used
-USEMODULE += ng_netif
-USEMODULE += ng_sixlowpan_frag
-USEMODULE += ng_ipv6
+USEMODULE += ng_ipv6_default
+USEMODULE += auto_init_ng_netif
+USEMODULE += ng_nomac
+USEMODULE += ng_at86rf231
+USEMODULE += ng_icmpv6_echo
 USEMODULE += ng_udp
+USEMODULE += fib
 USEMODULE += ng_pktdump
 USEMODULE += uart0
 USEMODULE += shell
@@ -27,3 +30,4 @@ CFLAGS += -DDEVELHELP
 QUIET ?= 1
 
 include $(RIOTBASE)/Makefile.include
+include $(RIOTBASE)/dist/testbed-support/Makefile.iot-lab
```
and call
```
BOARD=iot-lab_M3 make iotlab-exp iotlab-term
```
from `/examples/ng_networking`.

authmillenon, 2015-05-27:
I think I encountered this too and it is related to the ZEP thing I mentioned. But did you encounter the source address thingie (#2961)?

authmillenon, 2015-05-27:
+ I can't test your set-up since I still don't have an IoT-Lab account :(

OlegHahm, 2015-05-27:
> I can't test your set-up since I still don't have an IoT-Lab account :(
https://www.iot-lab.info/testbed/signup.php

OlegHahm, 2015-05-27:
> I think I encountered this too and it is related to the ZEP thing I mentioned. 

How is ZEP involved here?

OlegHahm, 2015-05-27:
> But did you encounter the source address thingie (#2961)?

Will check.

authmillenon, 2015-05-27:
> How is ZEP involved here?

I had a pings of death, when testing #2627.

OlegHahm, 2015-05-27:
Ah, ok, but the bug is then probably not related to ZEP.

authmillenon, 2015-05-27:
>> I can't test your set-up since I still don't have an IoT-Lab account :(
>
> https://www.iot-lab.info/testbed/signup.php

Nothing happens when I hit the sign-up button…

authmillenon, 2015-05-27:
(I reported that ages ago)

authmillenon, 2015-05-27:
Oh wait it shows "error" on the top.

OlegHahm, 2015-05-27:
I sent a mail to the list.

authmillenon, 2015-05-27:
> Ah, ok, but the bug is then probably not related to ZEP.

Likely, but would not count on that :D

OlegHahm, 2015-05-27:
Ok, let's rephrase there is at least one bug that's not related to ZEP.

authmillenon, 2015-05-27:
What about #2961?

OlegHahm, 2015-05-27:
Had to fix the other stuff first, now I will investigate.

OlegHahm, 2015-05-27:
See https://www.cloudshark.org/captures/158abcbaef10 after the last echo request, the other node crashed.

authmillenon, 2015-05-27:
I don't even can confirm reachability between two nodes :/

OlegHahm, 2015-05-27:
Which nodes? What's happening?

authmillenon, 2015-05-27:
```
1432727553.079962;m3-164;> ifconfig
1432727553.080886;m3-163;> ifconfig
1432727553.081615;m3-164;Iface  8   HWaddr: 7e:02  Channel: 17  NID: 0x23  TX-Power: 0dBm  State: TX
1432727553.081874;m3-165;> ifconfig
1432727553.082700;m3-163;Iface  8   HWaddr: 89:22  Channel: 17  NID: 0x23  TX-Power: 0dBm  State: TX
1432727553.084304;m3-164;           Long HWaddr: 36:32:48:33:46:dc:7e:02
1432727553.085037;m3-164;           AUTOACK  6LO
1432727553.085219;m3-164;           Source address length: 2
1432727553.085582;m3-165;Iface  8   HWaddr: 87:26  Channel: 17  NID: 0x23  TX-Power: 0dBm  State: IDLE
1432727553.085800;m3-165;           Long HWaddr: 36:32:48:33:46:d8:87:26
1432727553.085934;m3-165;           AUTOACK  6LO
1432727553.086096;m3-165;           Source address length: 2
1432727553.086521;m3-162;> ifconfig
1432727553.086705;m3-162;Iface  8   HWaddr: 8a:36  Channel: 17  NID: 0x23  TX-Power: 0dBm  State: TX
1432727553.087076;m3-162;           Long HWaddr: 36:32:48:33:46:d5:8a:36
1432727553.087242;m3-162;           AUTOACK  6LO
1432727553.087415;m3-162;           Source address length: 2
1432727553.087627;m3-162;           inet6 addr: ff02::1/128  scope: local [multicast]
1432727553.087954;m3-163;           Long HWaddr: 36:32:48:33:46:de:89:22
1432727553.088148;m3-163;           AUTOACK  6LO
1432727553.088813;m3-163;           Source address length: 2
1432727553.089194;m3-164;           inet6 addr: ff02::1/128  scope: local [multicast]
1432727553.089705;m3-165;           inet6 addr: ff02::1/128  scope: local [multicast]
1432727553.090044;m3-162;           inet6 addr: fe80::ff:fe00:8a36/64  scope: local
1432727553.090463;m3-163;           inet6 addr: ff02::1/128  scope: local [multicast]
1432727553.090664;m3-163;           inet6 addr: fe80::ff:fe00:8922/64  scope: local
1432727553.091225;m3-164;           inet6 addr: fe80::ff:fe00:7e02/64  scope: local
1432727553.091422;m3-164;           inet6 addr: ff02::1:ff00:7e02/128  scope: local [multicast]
1432727553.091631;m3-164;          
1432727553.092007;m3-165;           inet6 addr: fe80::ff:fe00:8726/64  scope: local
1432727553.092381;m3-165;           inet6 addr: ff02::1:ff00:8726/128  scope: local [multicast]
1432727553.092620;m3-165;          
1432727553.093039;m3-162;           inet6 addr: ff02::1:ff00:8a36/128  scope: local [multicast]
1432727553.093223;m3-162;          
1432727553.093625;m3-163;           inet6 addr: ff02::1:ff00:8922/128  scope: local [multicast]
1432727553.093805;m3-163;          
m3-165;txtsnd 8 7e:02 test
1432727581.675931;m3-165;> txtsnd 8 7e:02 test
1432727581.676860;m3-165;pktbuf: allocated (new_pktsnip = 0) of size 17
1432727581.678179;m3-165;pktbuf: allocated (new_pktsnip->data = 0x8000291) of size 4
1432727581.678891;m3-165;pktbuf: copying 4 byte from 0x200008af to 0x20001a18
1432727581.679866;m3-165;pktbuf: prepended new_pktsnip to pkt
1432727581.681817;m3-165;pktbuf: summary of new snip 0x200019fc: next = 0, data = 0x20001a18, size = 4, type = 0, users: 1
1432727581.683631;m3-165;pktbuf: allocated (new_pktsnip = 0x200019fc) of size 17
1432727581.684766;m3-165;pktbuf: allocated (new_pktsnip->data = 0x20001a18) of size 9
1432727581.685683;m3-165;pktbuf: prepended new_pktsnip to pkt
1432727581.687639;m3-165;pktbuf: summary of new snip 0x20001a24: next = 0x200019fc, data = 0x20001a40, size = 9, type = -1, users: 1
1432727581.688764;m3-165;Cannot debug, stack too small
1432727581.689018;m3-165;Cannot debug, stack too small
1432727581.689589;m3-165;Cannot debug, stack too small
1432727581.689786;m3-165;Cannot debug, stack too small
1432727581.690636;m3-165;Cannot debug, stack too small
1432727581.691623;m3-165;Cannot debug, stack too small
e1432727759.943761;Stopping
```

with current master (fe462fe9), #2895, #2952, #3084 and #3089 merged

authmillenon, 2015-05-27:
(and the modifications you described above)

OlegHahm, 2015-05-27:
Have you tried with default or ng_networking example?

authmillenon, 2015-05-28:
The ng_networking example

OlegHahm, 2015-05-28:
Ok, was just checking, because I tried yesterday evening with `default` and applied the same changes (at least I think I did) and didn't reach the neighboring nodes either. Probably missed something there.

With ng_networking I can confirm that the bug seems not to be appear on native (and valgrind is also pretty silent) and seems also to disappear on IoT-LAB with debugging enabled. My conclusion is, that it might be timing related.

authmillenon, 2015-05-28:
Would also be my guess especially since the packet buffer seems to handle things quite nicely this time and the crash seems to be quite arbitrary from a network stack point of view. Both `ng_icmpv6_echo` and `ng_ndp` make use of `vtimer`.

mixania, 2015-06-05:
I know this is not the best place for my question but is there a way to run RIOT on a Teensy board?

authmillenon, 2015-06-05:
Can you please post this question to devel@riot-os.org. That is a better place for that question ;-)

OlegHahm, 2015-08-17:
Seemed to be fixed.

closed 2015-08-17 by OlegHahm