RIOT-OS/RIOT #3086: Max. packet length for AT86RF2XX

PeterKietzmann, 2015-05-27:
In [`ng_sixlowpan`](https://github.com/RIOT-OS/RIOT/blob/master/sys/net/network_layer/ng_sixlowpan/ng_sixlowpan.c#L186) the max. packet size `NETCONF_OPT_MAX_PACKET_SIZE` is requested from the hardware device which correlates with the max. fragment size of a 6LoWPAN packet. It is the basis for 6LoWPAN fragmentation, if payloads are too long. In the [art86rfxx netdev-driver](https://github.com/RIOT-OS/RIOT/blob/master/drivers/ng_at86rf2xx/ng_at86rf2xx_netdev.c#L427) the device returns the value [`NG_AT86RF2XX_MAX_PKT_LENGTH    (127)`](https://github.com/RIOT-OS/RIOT/blob/master/drivers/include/ng_at86rf2xx.h#L44). In the drivers [`_send` function](https://github.com/RIOT-OS/RIOT/blob/master/drivers/ng_at86rf2xx/ng_at86rf2xx_netdev.c#L228) it is checked if the transceivers max. packet length __plus__ `FCS` check sequence __plus__ 802.15.4 header length is greater than the transceivers max. fifo size.
What I'm saying is: Shouldn't we return the max. packet size __minus__ link layer header length to the 6LoWPAN fragmentation? Or do I get confused somewhere between fragmentation, header lengths and the transceivers fifo size?

authmillenon, 2015-05-27:
At least `xbee` seems to do it the way you describe it, so I guess you are right :D

PeterKietzmann, 2015-07-22:
For the record:  #3480

PeterKietzmann, 2015-07-23:
@OlegHahm I quickly adapted the issue-discription. Can you please read it and check if one understands the problem?

gebart, 2015-07-23:
I understand the edited problem statement.

OlegHahm, 2015-07-23:
@PeterKietzmann, yes, I think it's comprehensible.

Btw. I discussed yesterday shortly with @authmillenon and we concluded that basically everything except the destination address length should be known to the link layer/transceiver driver.

PeterKietzmann, 2015-07-23:
Among others, the function [`_make_data_frame_hdr`](https://github.com/RIOT-OS/RIOT/blob/master/drivers/ng_at86rf2xx/ng_at86rf2xx_netdev.c#L36) checks for the header flags `NG_NETIF_HDR_FLAGS_BROADCAST` and/or `NG_NETIF_HDR_FLAGS_MULTICAST`. Is this known by the driver?

OlegHahm, 2016-02-24:
@authmillenon, how is the situation in the new (netdev2) driver version?

authmillenon, 2016-02-24:
Nothing yet, but since netdev2 centralizes this. Doing this should be quite easy.

authmillenon, 2016-02-24:
But I would vote against doing it in #4646.

OlegHahm, 2016-02-24:
> But I would vote against doing it in #4646.

Agreed!

kYc0o, 2016-08-04:
We leave this issue open until is fully solved?

miri64, 2016-08-04:
It's not solved yet, so yes ;-)

miri64, 2016-10-31:
Again postponed.

miri64, 2016-11-10:
Mhhh… this is more of an enhancement by now, not an issue => postponed.

miri64, 2020-02-11:
@jia200x might be of interest for you?

jia200x, 2020-02-11:
yes! Thanks!

maribu, 2022-09-16:
ping?

maribu, 2023-05-17:
If I understand correctly this goes in the same direction as https://github.com/RIOT-OS/RIOT/issues/19132.

Normally, it is common procedure to close the new issues as duplicates of the old. In this case, I think the new issue is more general and this one a subset of the new issue. Hence, I'd like to close the old one here instead.

Please reopen if you disagree.

closed 2023-05-17 by maribu