RIOT-OS/RIOT #3088: pktbuf: compiler warning on atomic_dec()

authmillenon, 2015-05-27:
Compiling `ng_pktbuf` since 6821350 yields a compiler warning.

```
/home/martine/Repositories/RIOT-OS/RIOT/sys/net/crosslayer/ng_pktbuf/ng_pktbuf.c: In function 'ng_pktbuf_start_write':
/home/martine/Repositories/RIOT-OS/RIOT/sys/net/crosslayer/ng_pktbuf/ng_pktbuf.c:151:20: warning: passing argument 1 of 'atomic_dec' from incompatible pointer type
         atomic_dec(&pkt->users);
                    ^
In file included from /home/martine/Repositories/RIOT-OS/RIOT/core/include/mutex.h:25:0,
                 from /home/martine/Repositories/RIOT-OS/RIOT/sys/net/crosslayer/ng_pktbuf/ng_pktbuf.c:25:
/home/martine/Repositories/RIOT-OS/RIOT/core/include/atomic.h:81:19: note: expected 'struct atomic_int_t *' but argument is of type 'unsigned int *'
 static inline int atomic_dec(atomic_int_t *var)
                   ^
```

However, `atomic_int_t`'s member is signed while `pkt->users` is unsigned, which could make simple casting problematic (not to mention that `atomic_int_t::value` is volatile).

OlegHahm, 2015-05-27:
Why didn't building the pktbuf unittest in Travis reveal that?

authmillenon, 2015-05-27:
Because travis does not output warnings, afaik.

authmillenon, 2015-05-27:
(This was building for `iot-lab_M3`)

OlegHahm, 2015-05-27:
> Because travis does not output warnings, afaik.

I always thought it would (and it definitely should), but I have no insights there.

authmillenon, 2015-05-27:
For unittests on native it does (but there the warning level seems not to be configured that strict), but not for the general build-tests

OlegHahm, 2015-05-27:
damn

Kijewski, 2015-05-27:
And yet someone wanted to close #1121 just recently. ;)

authmillenon, 2015-05-27:
#1121 doesn't fix the problem as long as there ARE warnings (which noone seems to care to fix)… The warnings used to be shown for all buildtests, but someone's python script made that basically impossible without rewriting the whole thing</passiveAggressiveness>

OlegHahm, 2015-05-27:
> which noone seems to care to fix

I fix warnings whenever I see them, but if our CI server let them pass, it's a bit "sisyphussy".

gebart, 2015-05-27:
sorry about this mistake. Is there a problem converting `users` to an `atomic_int_t` and making it signed?

authmillenon, 2015-05-27:
Since `users` is never supposed to go below 0 I'd rather don't want to make it signed. Why is there now `atomic_unsigned_t`?

gebart, 2015-05-27:
@authmillenon See the discussion between me, @OlegHahm and @haukepetersen at https://github.com/RIOT-OS/RIOT/issues/2302#issuecomment-69900667

authmillenon, 2015-05-27:
I just realized that the atomic operation there is a relic from a previous, mutex avoiding implementation. However, since the operation is now in a mutex, a normal `pkt->users--` should suffice.

BytesGalore, 2015-06-02:
+1 for `pkt->users--` 
is there already a PR for this?

authmillenon, 2015-06-02:
Yes #3092

PR #3092 (2015-06-02): ng_pktbuf: Some fixes
Fixes #3088 and also some weird artifacts I discovered when writing #3089.
closed 2015-06-02 by authmillenon