RIOT-OS/RIOT #3109: periph/random: random_read should return unsigned int

gebart, 2015-05-29:
https://github.com/RIOT-OS/RIOT/pull/3101#discussion_r31214572

The documentation for `periph/random.h` says:

>Reads num or less bytes of randomness from the source, will block until random data is available.
>
>Parameters
>    [in]	buf	destination buffer to write the bytes to
>    [in]	num	number of bytes to get from device, only values >0 are valid
>
>Returns
>    the number of bytes written to buf 
>    0 on error 

so we should not need to have a signed int for return type.

OlegHahm, 2015-12-04:
@PeterKietzmann, @gebart, untag for this release?

PeterKietzmann, 2015-12-04:
Yes, let's untag it for this release **but** come to a conclusion here. I think that I am pro API change, even if adapting the documentation might be easier. Any other opinions?

thomaseichinger, 2015-12-04:
I'd support changing the return type for consistency.

PR #4757 (2016-02-14): drivers/periph: renamed random and more
The random peripheral driver interface needed an overhaul to comply with the other modernized peripheral interfaces:
- renamed it to 'hrng' -> many people complained about the old name...
- removed power on/off functions (as the device should be powered off by default and only powered on while used)
- none of our implementations have any valid error case, so no need for a return value on 'read'

This also fixes #3109

Travis won't be happy, as the path for the `test/pkg_micro-ecc` is still untouched.. .Would somebody mind to tell me the fasted way to cha [... cut]
closed 2016-02-14 by gebart