RIOT-OS/RIOT #3125: ng_netdev: TX feedback

daniel-k, 2015-05-30:
The new netdev API seems to lack any kind of transmission feedback. In the former days when transmitting data over a 802.15.4 transceiver by 

```c
netdev_802154_tx_status_t (* transmit)(netdev_t *dev);
```

directly returned the result of a transmission. The current implementation in `ng_netdev.h` only provides:
```c
    /* [...]
     * @return              number of bytes that were actually send out
     * @return              -ENODEV if @p dev is invalid
     * @return              -ENOMSG if pkt is invalid
     * @return              -EOVERFLOW if the payload size of @p pkt exceeds the
     *                      payload size that can be handled by the device
     */
    int (*send_data)(ng_netdev_t *dev, ng_pktsnip_t *pkt);
```

I see that netdev shall be a general API not only for radio transceivers, but is there already a replacement for TX feedback? Maybe I didn't search carefully enough?

Still the old implementation has some drawbacks: e.g. when waiting for an ACK the transmit function needs to be blocking in order to return the TX feedback. 

With `ng_netdev` this could be implemented (non-blocking) by adding new events to `ng_netdev_event_t` I guess.

```c
typedef enum {
    NETDEV_EVENT_RX_STARTED     = 0x0001,   /**< started to receive a packet */
    NETDEV_EVENT_RX_COMPLETE    = 0x0002,   /**< finished receiving a packet */
    NETDEV_EVENT_TX_STARTED     = 0x0004,   /**< started to transfer a packet */
    NETDEV_EVENT_TX_COMPLETE    = 0x0008,   /**< finished transferring packet */
    /* expand this list if needed */
    NETDEV_EVENT_TX_NOACK = 0x4242; /**< TX successful, but no ACK received */
} ng_netdev_event_t;
```

daniel-k, 2015-09-24:
Just found this again. See #3729, as this is my reference implementation for at86rf2xx.

OlegHahm, 2016-03-09:
Oops, I think we overlooked this issue while discussing here: https://github.com/RIOT-OS/RIOT/pull/4648#issuecomment-177008623

miri64, 2016-10-17:
Wasn't this fixed (and opened a whole new can of worms -> #5486)?

daniel-k, 2016-11-02:
My according PR #3729 has already been merged a while ago, so I think we can close this issue. As far as I can tell, this is unrelated to #5486.

PeterKietzmann, 2016-11-02:
Thanks for the update @daniel-k!

closed 2016-11-02 by PeterKietzmann