RIOT-OS/RIOT #3139: kinetis: hwtimer loses ticks when referenced against rtt ticks

gebart, 2015-06-02:
The hwtimer_now function loses something in the order of 130 ticks in 60 seconds. The hwtimer is using the LPTMR module which runs off the RTC crystal at 32768 Hz, so the loss is 130/32768 approx 4 ms per second. These numbers are from a test program where I compare the RTT seconds and sub tick counter registers against the time reported by hwtimer_now().

I am working on a refactor of the hwtimer module in kinetis_common.

gebart, 2015-06-02:
I had some patches when achieving that number. Without my WIP patches for the hwtimer, the loss is bigger, 240 ticks lost in 60 seconds against RTT.

kaspar030, 2015-06-02:
does kinetis implement the periph/timer interface? If not, you should focus on moving to that.

gebart, 2015-06-02:
@kaspar030 The problem is that we are using another timer hardware (PIT) for the periph/timer interface which allows for higher clock frequencies and shorter periods, but it stops when the CPU goes to sleep, so it is worthless for scheduling periodic tasks that are farther away than a few µs or ms. The LPTMR only exists as a single channel timer in the CPU and we want to use that for scheduling etc because it keeps running when the CPU is put in low power modes.

Do you have any suggestions for a good driver design in this case?

gebart, 2015-09-24:
closing because hwtimer has been removed. The theoretical problems with using PIT and LPTMR still has to be solved for xtimer before LPM can be properly implemented though.

closed 2015-09-24 by gebart