RIOT-OS/RIOT #3169: ng_at86rf2xx: CCA when in Extended Mode

daniel-k, 2015-06-05:
At the moment the at86rf2xx driver issues a manual CCA no matter in which state the transceiver is. Additionally, only the Extended Mode (RX_AACK_ON) is implemented. However the datasheet has some quite explicit sections:

p. 104
> It is not recommended to manually initiate an CCA measurement when using the Extended Operating Mode.

p. 105
> A CCA request should only be initiated in Basic Operating Mode receive states. Otherwise the radio transceiver generates an IRQ_4 (CCA_ED_DONE) and sets the register bit CCA_DONE = 1, even though no CCA measurement was performed.

p. 107
> If a CCA request is initiated in states others than RX_ON or RX_BUSY the PHY generates an IRQ_4 (CCA_ED_DONE) and sets the register bit CCA_DONE, however no CCA was carried out.

Did you test the behaviour in Extended Mode? The datasheet uses *should* and *recommended* so I guess CCA might also work in Extended Mode in some cases. What do you think?

kaspar030, 2015-12-18:
(too late for 2015.12)

haukepetersen, 2016-03-31:
@thomaseichinger, @OlegHahm: do we have any news on this?

OlegHahm, 2016-04-01:
I have started to work on this, but with low priority and since it is a new feature, it won't make it for the release.

thomaseichinger, 2016-04-02:
As far as I tested it in the past, triggering CCA requests in software while in extended operation modes never worked. This kind of contradicts p.107
> p. 107

>>If a CCA request is initiated in states others than RX_ON or RX_BUSY the PHY generates an IRQ_4 >>(CCA_ED_DONE) and sets the register bit CCA_DONE, however no CCA was carried out.

haukepetersen, 2016-04-20:
postponed

kYc0o, 2016-07-26:
And postponed again...

miri64, 2016-10-17:
at86rf2xx is not part of GNRC anymore.

miri64, 2016-11-04:
This is not a bug, so I'm postponing it.

OlegHahm, 2016-12-23:
What would be the "right" way to tackle this issue? Disabling the CCA netopt for the current implementation?

PeterKietzmann, 2017-01-10:
It seems no one is actively working on this. I will remove the Milestone completely which must not mean that it's not worth it working on such a great feature.

OlegHahm, 2017-01-14:
I still wonder what is the actual goal of this issue. Implementing basic mode for the transceiver or simply disable the feature for now?

OlegHahm, 2017-01-14:
I propose to go with #6355 first and work on a driver version for the basic operation mode of the at86rf2xx after the release.

OlegHahm, 2017-01-25:
Fixed with #6355 (no CCA in extended mode any more).

closed 2017-01-25 by OlegHahm