RIOT-OS/RIOT #3175: Native: build of vtimer_msg test fails on OS X

emmanuelsearch, 2015-06-08:
Build of vtimer_msg test fails on OS X with current master. Error is:

```
In file included from /RIOT/cpu/native/hwtimer_cpu.c:47:
In file included from /RIOT/RIOT/core/include/debug.h:34:
/RIOT/core/include/thread.h:138:14: error: conflicting types for 'thread_create'
kernel_pid_t thread_create(char *stack,
             ^
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.9.sdk/usr/include/mach/task.h:193:15: note: 
      previous declaration is here
kern_return_t thread_create
              ^
1 error generated.
```

josephnoir, 2015-06-09:
Same for me in the `thread_*` tests.
```
$ clang -v
OS X 10.10.4 (14E26a)
Apple LLVM version 6.1.0 (clang-602.0.53) (based on LLVM 3.6.0svn)
Target: x86_64-apple-darwin14.4.0
Thread model: posix
´´´

ximus, 2015-06-15:
This also happens when running the entire test suite `cd tests/unittests && make`
My output is slightly different (different location of task.h), but essentially the same error:

```
~/c/R/t/unittests (master) $ make
Configured with: --prefix=/Applications/Xcode.app/Contents/Developer/usr --with-gxx-include-dir=/usr/include/c++/4.2.1
Building application "unittests" for "native" with MCU "native".

touch /Users/me/code/RIOT/tests/unittests/main.c
"/Applications/Xcode.app/Contents/Developer/usr/bin/make" -C /Users/me/code/RIOT/boards/native
"/Applications/Xcode.app/Contents/Developer/usr/bin/make" -C /Users/me/code/RIOT/boards/native/drivers
"/Applications/Xcode.app/Contents/Developer/usr/bin/make" -C /Users/me/code/RIOT/core
"/Applications/Xcode.app/Contents/Developer/usr/bin/make" -C /Users/me/code/RIOT/cpu/native
In file included from /Users/me/code/RIOT/cpu/native/hwtimer_cpu.c:47:
In file included from /Users/me/code/RIOT/core/include/debug.h:34:
/Users/me/code/RIOT/core/include/thread.h:138:14: error: conflicting types for 'thread_create'
kernel_pid_t thread_create(char *stack,
             ^
/usr/include/mach/task.h:193:15: note: previous declaration is here
kern_return_t thread_create
              ^
1 error generated.
make[2]: *** [/Users/me/code/RIOT/tests/unittests/bin/native/cpu/hwtimer_cpu.o] Error 1
make[1]: *** [ALL--/Users/me/code/RIOT/cpu/native] Error 2
make: *** [all] Error 2
```

The issue is `cpu/native/hwtimer_cpu.c` on OSX includes `mach/mach.h` which includes `mach/mach_interface.h` which includes `mach/task.h` which defines `thread_create()` which conflicts with Riot's own `thread_create()`.

I see two solutions:
* Rename the conflicting method(s). Downside is `thread_create()` is a good and existing name.
* Rework the mach dependencies, which is possible. This works for the time being. I was able to get past the error with the following change:

```
diff --git a/cpu/native/hwtimer_cpu.c b/cpu/native/hwtimer_cpu.c
index 2dc9f1b..4fe0e17 100644
--- a/cpu/native/hwtimer_cpu.c
+++ b/cpu/native/hwtimer_cpu.c
@@ -23,7 +23,8 @@

 #ifdef __MACH__
 #include <mach/clock.h>
-#include <mach/mach.h>
+#include <mach/mach_init.h>
+#include <mach/mach_port.h>
 #endif

 #include <time.h>
```

emmanuelsearch, 2015-07-02:
Seems to work. But I still get quite some warnings:

```
make -B clean all
Configured with: --prefix=/Applications/Xcode.app/Contents/Developer/usr --with-gxx-include-dir=/usr/include/c++/4.2.1
Building application "vtimer_msg" for "native" with MCU "native".

"/Applications/Xcode.app/Contents/Developer/usr/bin/make" -C /Users/emmanuelB/Documents/work/CODING/RIOT/RIOT/boards/native
"/Applications/Xcode.app/Contents/Developer/usr/bin/make" -C /Users/emmanuelB/Documents/work/CODING/RIOT/RIOT/boards/native/drivers
"/Applications/Xcode.app/Contents/Developer/usr/bin/make" -C /Users/emmanuelB/Documents/work/CODING/RIOT/RIOT/core
"/Applications/Xcode.app/Contents/Developer/usr/bin/make" -C /Users/emmanuelB/Documents/work/CODING/RIOT/RIOT/cpu/native
/Users/emmanuelB/Documents/work/CODING/RIOT/RIOT/cpu/native/hwtimer_cpu.c:280:5: warning: 
      implicit declaration of function 'host_get_clock_service' is invalid in
      C99 [-Wimplicit-function-declaration]
    host_get_clock_service(mach_host_self(), SYSTEM_CLOCK, &cclock);
    ^
1 warning generated.
/Users/emmanuelB/Documents/work/CODING/RIOT/RIOT/cpu/native/irq_cpu.c:324:5: warning: 
      'makecontext' is deprecated: first deprecated in OS X 10.6
      [-Wdeprecated-declarations]
    makecontext(&native_isr_context, native_irq_handler, 0);
    ^
/usr/include/ucontext.h:38:6: note: 'makecontext' has been explicitly marked
      deprecated here
void makecontext(ucontext_t *, void (*)(), int, ...) __OSX_AVAILABLE_BUT...
     ^
/Users/emmanuelB/Documents/work/CODING/RIOT/RIOT/cpu/native/irq_cpu.c:486:9: warning: 
      'getcontext' is deprecated: first deprecated in OS X 10.6
      [-Wdeprecated-declarations]
    if (getcontext(&native_isr_context) == -1) {
        ^
/usr/include/ucontext.h:37:6: note: 'getcontext' has been explicitly marked
      deprecated here
int  getcontext(ucontext_t *) __OSX_AVAILABLE_BUT_DEPRECATED(__MAC_10_5...
     ^
/Users/emmanuelB/Documents/work/CODING/RIOT/RIOT/cpu/native/irq_cpu.c:504:5: warning: 
      'makecontext' is deprecated: first deprecated in OS X 10.6
      [-Wdeprecated-declarations]
    makecontext(&native_isr_context, native_irq_handler, 0);
    ^
/usr/include/ucontext.h:38:6: note: 'makecontext' has been explicitly marked
      deprecated here
void makecontext(ucontext_t *, void (*)(), int, ...) __OSX_AVAILABLE_BUT...
     ^
3 warnings generated.
/Users/emmanuelB/Documents/work/CODING/RIOT/RIOT/cpu/native/native_cpu.c:118:9: warning: 
      'getcontext' is deprecated: first deprecated in OS X 10.6
      [-Wdeprecated-declarations]
    if (getcontext(p) == -1) {
        ^
/usr/include/ucontext.h:37:6: note: 'getcontext' has been explicitly marked
      deprecated here
int  getcontext(ucontext_t *) __OSX_AVAILABLE_BUT_DEPRECATED(__MAC_10_5...
     ^
/Users/emmanuelB/Documents/work/CODING/RIOT/RIOT/cpu/native/native_cpu.c:131:5: warning: 
      'makecontext' is deprecated: first deprecated in OS X 10.6
      [-Wdeprecated-declarations]
    makecontext(p, (void (*)(void)) task_func, 1, arg);
    ^
/usr/include/ucontext.h:38:6: note: 'makecontext' has been explicitly marked
      deprecated here
void makecontext(ucontext_t *, void (*)(), int, ...) __OSX_AVAILABLE_BUT...
     ^
/Users/emmanuelB/Documents/work/CODING/RIOT/RIOT/cpu/native/native_cpu.c:153:9: warning: 
      'setcontext' is deprecated: first deprecated in OS X 10.6
      [-Wdeprecated-declarations]
    if (setcontext(ctx) == -1) {
        ^
/usr/include/ucontext.h:39:6: note: 'setcontext' has been explicitly marked
      deprecated here
[... 69 lines cut ...]
        makecontext(&native_isr_context, native_irq_handler, 0);
        ^
/usr/include/ucontext.h:38:6: note: 'makecontext' has been explicitly marked
      deprecated here
void makecontext(ucontext_t *, void (*)(), int, ...) __OSX_AVAILABLE_BUT...
     ^
/Users/emmanuelB/Documents/work/CODING/RIOT/RIOT/cpu/native/syscalls.c:129:13: warning: 
      'swapcontext' is deprecated: first deprecated in OS X 10.6
      [-Wdeprecated-declarations]
        if (swapcontext(_native_cur_ctx, &native_isr_context) == -1) {
            ^
/usr/include/ucontext.h:40:6: note: 'swapcontext' has been explicitly marked
      deprecated here
int  swapcontext(ucontext_t * __restrict, const ucontext_t * __restrict...
     ^
2 warnings generated.
"/Applications/Xcode.app/Contents/Developer/usr/bin/make" -C /Users/emmanuelB/Documents/work/CODING/RIOT/RIOT/cpu/native/periph
"/Applications/Xcode.app/Contents/Developer/usr/bin/make" -C /Users/emmanuelB/Documents/work/CODING/RIOT/RIOT/drivers
"/Applications/Xcode.app/Contents/Developer/usr/bin/make" -C /Users/emmanuelB/Documents/work/CODING/RIOT/RIOT/sys
"/Applications/Xcode.app/Contents/Developer/usr/bin/make" -C /Users/emmanuelB/Documents/work/CODING/RIOT/RIOT/sys/auto_init
"/Applications/Xcode.app/Contents/Developer/usr/bin/make" -C /Users/emmanuelB/Documents/work/CODING/RIOT/RIOT/sys/timex
"/Applications/Xcode.app/Contents/Developer/usr/bin/make" -C /Users/emmanuelB/Documents/work/CODING/RIOT/RIOT/sys/vtimer
__TEXT	__DATA	__OBJC	others	dec	hex
36864	1544192	0	20480	1601536	187000
```

kaspar030, 2015-07-09:
Fixed for me in master. @thomaseichinger @emmanuelsearch could you confirm and close?

emmanuelsearch, 2015-07-09:
builds on master indeed.

closed 2015-07-09 by emmanuelsearch