RIOT-OS/RIOT #3187: Atmega2560: shell and/or UART_RX dosn't work correctly?

PeterKietzmann, 2015-06-11:
I tried to use the shell on the arduino-mega2560 without success. Is this known or am I doing something wrong? Then I tried the `periph_uart_int` test and I get stuck somewhere after [Setting up the buffers](https://github.com/RIOT-OS/RIOT/blob/master/tests/periph_uart_int/main.c#L100). Can anyone with more experience on the atmega2560 confirm this? Any ideas?

kaspar030, 2015-06-11:
On 06/11/15 10:45, Peter Kietzmann wrote:
> I tried to use the shell on the arduino-mega2560 without success. Is
> this known or am I doing something wrong? Then I tried the
> |periph_uart_int| test and I get stuck somewhere after Setting up the [...]

In my remove_uart0 branch, I've adapted arduino-mega2560 to use the new
uart_stdio abstraction. If you don't use uart0 (like in that branches
examples/default), the shell should work.

PeterKietzmann, 2015-06-11:
Did you test anything else than the `ps` command in the default example? I don't know it it's the length of a shell command but other commands than `ps`, e.g. `reboot`or some in my own test, don't really work. Sometimes the board gets stuck, sometimes there are just some snips of a command that were recognized. As an example:
```
2015-06-11 12:17:59,046 - INFO # > help
2015-06-11 12:17:59,054 - INFO # Command              Description
2015-06-11 12:17:59,066 - INFO # ---------------------------------------
2015-06-11 12:17:59,075 - INFO # reboot               Reboot the node
2015-06-11 12:17:59,095 - INFO # ps                   Prints information about running threads.
reboot

2015-06-11 12:18:14,152 - INFO # > re> > > )reb> > )
```

kaspar030, 2015-06-11:
> Did you test anything else than the |ps| command in the default example?

Actually, no. I don't have a board, so I adapted the port blind and just
did a quick test at last NSTF's hackathon...

N8Fear, 2015-06-11:
Sadly I still don't have enough time to do work on RIOT. 
@PeterKietzmann : Concerning your issues with `periph_uart_int`: this test uses `115200` Baud per default which seems to be to much for the Atmega2560 to handle. The most I got out of it without getting funny characters is `38400`. There are other issues with this test, too: `vtimer_usleep(2000*1000)` overflows the max value for the timers (Atmega2560 has only 8 and 16 bit timers). I wanted to implement a software solution (by using the overflow interrupt) I never really got to finish that one (it's potentially racy too).
There was another issue with the timers that status is unknown to me (the bug is still open: https://github.com/RIOT-OS/RIOT/issues/1811).
Concerning the corruption you get in the default example I can't reproduce it. Which Baudrate do you use here?
And last but not least the issue concerning the `reboot` command: the Atmega2560 doesn't have a native way to get a software reboot, so the only real way to get that functionality is by enabling the watchdog timer and let it time out. This way can lead to another issue (the "board gets stuck" description fits that one quite well): some versions of the bootloader don't disable the watchdog timer which causes the board to get stuck in a reset cycle (Here is some more info on that one: https://code.google.com/p/arduino/issues/detail?id=181). I guess the only way around that one is to update your device's bootloader (which requires either an ICE or a AVR Dragon).

PeterKietzmann, 2015-06-12:
@N8Fear thanks for your response! I totally overlooked the used baudrate in the test. I agree that 115200 baud is too fast for the atmega2560. Regarding the timer and reboot things it is also good to know. Still I think my proposed reboot issue was related to the UART input I gave.
Currently I also don't have time to go into too much details here. Would you say that #3186 and #1811 describe the same bug?

N8Fear, 2015-06-12:
I think it's quite possible at least. As stated in https://github.com/RIOT-OS/RIOT/issues/1811#issuecomment-60721330 I traced the issue to some clobbered mutex in `hwtimer_wait(..)`. I'm not sure what parts of the related code base changed in the meantime, but it's possible at least. IIRC I was able to prevent this issues by adding a short wait (or a printf to get a minimal delay). Maybe you could test if this makes the hwtimer related stuff work normal for you (which wouldn't be a fix but just a part of diagnosis).

PeterKietzmann, 2015-06-12:
Honestly I don't have time for this at the moment. The issues appeared as a side affect while trying to test the GPIO implementation #2799. I just wanted to report them so they won't be forgotten. If anyone has time and knowledge I would be happy if these bugs are fixed. Otherwise I'll look at it again at any time... Do you know if  #3186 and #1811 describe the same bug?

N8Fear, 2015-06-12:
As stated above: I don't know but I think it's very likely that both are the same.

PeterKietzmann, 2015-06-12:
Ups, I over-read the first sentence. Sorry! Thanks for the answer again.

ReneHerthel, 2015-11-03:
@PeterKietzmann I testet it for device UART0 and it works. Then I enabled device UART1 and I got following error message: 

>RIOT/cpu/atmega2560/periph/uart.c:214:12: error: ‘UART_2’ undeclared (first use in this function)
config[UART_2].rx_cb(config[UART_2].arg, UART0_DATA_REGISTER);

ReneHerthel, 2015-11-03:
@PeterKietzmann The [PR](https://github.com/RIOT-OS/RIOT/pull/4218) of my solution for this :)

ReneHerthel, 2015-11-07:
@PeterKietzmann This should work too now.

PeterKietzmann, 2015-11-09:
Fixed by #4218

closed 2015-11-09 by PeterKietzmann