RIOT-OS/RIOT #3195: sys/crypto: remove our Twofish implementation

Kijewski, 2015-06-12:
I do not know if there is a "IoT optimized" implementation of Twofish, but our implementation is unreasonably big for IoT applications and relies on malloc. The [reference implementation](https://www.schneier.com/twofish-download.html) looks much better at the first glance. 

Our twofish.c is unused and IMO unusable. Better drop it before it hurts someone ...

Kijewski, 2015-06-12:
Also remove skipjack. WTF is skipjack?

And RC5. RC5 is patented and cannot be used freely.

authmillenon, 2015-06-14:
According to the log they were imported by @haukepetersen in 05419a5 so I assign him.

haukepetersen, 2015-06-16:
The actual import was done by @zkasmi. I just pulled them in from some branch as far as I remember. I have no opinion about the twofish implementation - if it does not live up to our quality standards and if nobody is really using it then I would be also in favor of dropping support (at least for now).

Regarding Twofish: feel free to PR the reference implementation to replace the existing one!

OlegHahm, 2015-06-23:
I think these implementations originate from http://www.mi.fu-berlin.de/inf/groups/ag-tech/publications-old/1__resources/SNA-205.pdf

LudwigOrtmann, 2015-07-11:
I'm with @haukepetersen .

PR #6370 (2017-01-17): sys: crypto: remove twofish and rc5
Fixes #3195.
closed 2017-01-17 by miri64