RIOT-OS/RIOT #3198: examples: ng_UDP-6LoWPAN example does not compile on OS X

emmanuelsearch, 2015-06-12:
When compiling the example in OS X, I get the warning and the error below:

```
/RIOT/examples/ng_networking/udp.c:88:48: warning: format specifies type 'int' but the argument has
      type 'size_t' (aka 'unsigned long') [-Wformat]
    printf("Success: send %i byte to %s:%u\n", payload->size, addr_str, tmp);
                          ~~                   ^~~~~~~~~~~~~
                          %zu
1 warning generated.

"/Applications/Xcode.app/Contents/Developer/usr/bin/make" -C /RIOT/cpu/native
In file included from RIOT/cpu/native/hwtimer_cpu.c:47:
In file included from /RIOT/core/include/debug.h:34:
/RIOT/core/include/thread.h:138:14: error: conflicting types for 'thread_create'
kernel_pid_t thread_create(char *stack,
             ^
/usr/include/mach/task.h:193:15: note: previous declaration is here
kern_return_t thread_create
              ^
1 error generated.
```

Toolchain info:
OS X 10.10.3
$ gcc --version
Configured with: --prefix=/Applications/Xcode.app/Contents/Developer/usr --with-gxx-include-dir=/usr/include/c++/4.2.1
Apple LLVM version 7.0.0 (clang-700.0.53)
Target: x86_64-apple-darwin14.3.0
Thread model: posix

Kijewski, 2015-06-12:
Please test if adding this guard immediately before `kernel_pid_t thread_create(` in core/include/thread.h fixes the name clash for you:
```c
#if defined(__MACH__)
#   define thread_create riot_thread_create
#endif
```

emmanuelsearch, 2015-06-13:
mmmh. Here's what I get now with the fix you proposed:

```
$ make -B clean all
Configured with: --prefix=/Applications/Xcode.app/Contents/Developer/usr --with-gxx-include-dir=/usr/include/c++/4.2.1
Building application "ng_networking" for "native" with MCU "native".

/Users/emmanuelB/Documents/work/CODING/RIOT/RIOT/examples/ng_networking/udp.c:88:48: warning: format specifies type 'int' but the argument has
      type 'size_t' (aka 'unsigned long') [-Wformat]
    printf("Success: send %i byte to %s:%u\n", payload->size, addr_str, tmp);
                          ~~                   ^~~~~~~~~~~~~
                          %zu
1 warning generated.
"/Applications/Xcode.app/Contents/Developer/usr/bin/make" -C /Users/emmanuelB/Documents/work/CODING/RIOT/RIOT/boards/native
"/Applications/Xcode.app/Contents/Developer/usr/bin/make" -C /Users/emmanuelB/Documents/work/CODING/RIOT/RIOT/boards/native/drivers
"/Applications/Xcode.app/Contents/Developer/usr/bin/make" -C /Users/emmanuelB/Documents/work/CODING/RIOT/RIOT/core
"/Applications/Xcode.app/Contents/Developer/usr/bin/make" -C /Users/emmanuelB/Documents/work/CODING/RIOT/RIOT/cpu/native
/Users/emmanuelB/Documents/work/CODING/RIOT/RIOT/cpu/native/irq_cpu.c:324:5: warning: 'makecontext' is deprecated: first deprecated in OS X 10.6
      [-Wdeprecated-declarations]
    makecontext(&native_isr_context, native_irq_handler, 0);
    ^
/usr/include/ucontext.h:38:6: note: 'makecontext' has been explicitly marked deprecated here
void makecontext(ucontext_t *, void (*)(), int, ...) __OSX_AVAILABLE_BUT_DEPRECATED(__MAC_10_5, __MAC_10_6, __IPHONE_2_0, __IPHONE_2_0);
     ^
/Users/emmanuelB/Documents/work/CODING/RIOT/RIOT/cpu/native/irq_cpu.c:486:9: warning: 'getcontext' is deprecated: first deprecated in OS X 10.6
      [-Wdeprecated-declarations]
    if (getcontext(&native_isr_context) == -1) {
        ^
/usr/include/ucontext.h:37:6: note: 'getcontext' has been explicitly marked deprecated here
int  getcontext(ucontext_t *) __OSX_AVAILABLE_BUT_DEPRECATED(__MAC_10_5, __MAC_10_6, __IPHONE_2_0, __IPHONE_2_0);
     ^
/Users/emmanuelB/Documents/work/CODING/RIOT/RIOT/cpu/native/irq_cpu.c:504:5: warning: 'makecontext' is deprecated: first deprecated in OS X 10.6
      [-Wdeprecated-declarations]
    makecontext(&native_isr_context, native_irq_handler, 0);
    ^
/usr/include/ucontext.h:38:6: note: 'makecontext' has been explicitly marked deprecated here
void makecontext(ucontext_t *, void (*)(), int, ...) __OSX_AVAILABLE_BUT_DEPRECATED(__MAC_10_5, __MAC_10_6, __IPHONE_2_0, __IPHONE_2_0);
     ^
3 warnings generated.
/Users/emmanuelB/Documents/work/CODING/RIOT/RIOT/cpu/native/native_cpu.c:118:9: warning: 'getcontext' is deprecated: first deprecated in OS X 10.6
      [-Wdeprecated-declarations]
    if (getcontext(p) == -1) {
        ^
/usr/include/ucontext.h:37:6: note: 'getcontext' has been explicitly marked deprecated here
int  getcontext(ucontext_t *) __OSX_AVAILABLE_BUT_DEPRECATED(__MAC_10_5, __MAC_10_6, __IPHONE_2_0, __IPHONE_2_0);
     ^
/Users/emmanuelB/Documents/work/CODING/RIOT/RIOT/cpu/native/native_cpu.c:131:5: warning: 'makecontext' is deprecated: first deprecated in OS X
      10.6 [-Wdeprecated-declarations]
    makecontext(p, (void (*)(void)) task_func, 1, arg);
    ^
/usr/include/ucontext.h:38:6: note: 'makecontext' has been explicitly marked deprecated here
void makecontext(ucontext_t *, void (*)(), int, ...) __OSX_AVAILABLE_BUT_DEPRECATED(__MAC_10_5, __MAC_10_6, __IPHONE_2_0, __IPHONE_2_0);
     ^
/Users/emmanuelB/Documents/work/CODING/RIOT/RIOT/cpu/native/native_cpu.c:153:9: warning: 'setcontext' is deprecated: first deprecated in OS X 10.6
      [-Wdeprecated-declarations]
    if (setcontext(ctx) == -1) {
        ^
/usr/include/ucontext.h:39:6: note: 'setcontext' has been explicitly marked deprecated here
int  setcontext(const ucontext_t *) __OSX_AVAILABLE_BUT_DEPRECATED(__MAC_10_5, __MAC_10_6, __IPHONE_2_0, __IPHONE_2_0);
     ^
/Users/emmanuelB/Documents/work/CODING/RIOT/RIOT/cpu/native/native_cpu.c:174:9: warning: 'makecontext' is deprecated: first deprecated in OS X
      10.6 [-Wdeprecated-declarations]
        makecontext(&native_isr_context, isr_cpu_switch_context_exit, 0);
        ^
/usr/include/ucontext.h:38:6: note: 'makecontext' has been explicitly marked deprecated here
void makecontext(ucontext_t *, void (*)(), int, ...) __OSX_AVAILABLE_BUT_DEPRECATED(__MAC_10_5, __MAC_10_6, __IPHONE_2_0, __IPHONE_2_0);
     ^
/Users/emmanuelB/Documents/work/CODING/RIOT/RIOT/cpu/native/native_cpu.c:175:13: warning: 'setcontext' is deprecated: first deprecated in OS X
      10.6 [-Wdeprecated-declarations]
        if (setcontext(&native_isr_context) == -1) {
[... 97 lines cut ...]
    ^
/Users/emmanuelB/Documents/work/CODING/RIOT/RIOT/cpu/native/ng_net/dev_eth_tap.c:273:26: error: unknown type name 'ng_tapnet_t'
static void _sigio_child(ng_tapnet_t *dev)
                         ^
/Users/emmanuelB/Documents/work/CODING/RIOT/RIOT/cpu/native/ng_net/dev_eth_tap.c:276:10: error: use of undeclared identifier '_sigio_child_pid'
    if ((_sigio_child_pid = real_fork()) == -1) {
         ^
/Users/emmanuelB/Documents/work/CODING/RIOT/RIOT/cpu/native/ng_net/dev_eth_tap.c:279:9: error: use of undeclared identifier '_sigio_child_pid';
      did you mean '_sigio_child'?
    if (_sigio_child_pid > 0) {
        ^~~~~~~~~~~~~~~~
        _sigio_child
/Users/emmanuelB/Documents/work/CODING/RIOT/RIOT/cpu/native/ng_net/dev_eth_tap.c:257:5: note: '_sigio_child' declared here
[... cut ...]
      ('int (*)()' and 'int') is an extension [-Wpedantic]
    if (_sigio_child_pid > 0) {
        ~~~~~~~~~~~~~~~~ ^ ~
2 warnings and 14 errors generated.
make[3]: *** [/Users/emmanuelB/Documents/work/CODING/RIOT/RIOT/examples/ng_networking/bin/native/ng_nativenet/dev_eth_tap.o] Error 1
make[2]: *** [ALL--/Users/emmanuelB/Documents/work/CODING/RIOT/RIOT/cpu/native/ng_net] Error 2
make[1]: *** [ALL--/Users/emmanuelB/Documents/work/CODING/RIOT/RIOT/cpu/native] Error 2
make: *** [all] Error 2
```

thomaseichinger, 2015-06-16:
This problem was introduced with https://github.com/RIOT-OS/RIOT/pull/3144 it conflicts with `thread_create` declared in `mach_interfaces.h` included from `mach.h`. Removing `mach.h` from includes solves the compile problem for me.
@LudwigOrtmann can you still recall why it is included?

emmanuelsearch, 2015-06-16:
Just to be sure: which include in which file exactly are you saying we should remove?

thomaseichinger, 2015-06-17:
Sorry, I was not precise.
```diff
diff --git a/cpu/native/hwtimer_cpu.c b/cpu/native/hwtimer_cpu.c
index 2dc9f1b..3092ee8 100644
--- a/cpu/native/hwtimer_cpu.c
+++ b/cpu/native/hwtimer_cpu.c
@@ -23,7 +23,6 @@

 #ifdef __MACH__
 #include <mach/clock.h>
-#include <mach/mach.h>
 #endif

 #include <time.h>
```
is what I meant.

emmanuelsearch, 2015-06-18:
I had tried that (was the only place where mach.h occurred in the code base ;), but this is what I get now (still does not compile):

```
$ make -B clean all
Configured with: --prefix=/Applications/Xcode.app/Contents/Developer/usr --with-gxx-include-dir=/usr/include/c++/4.2.1
Building application "ng_networking" for "native" with MCU "native".

/Users/emmanuelB/Documents/work/CODING/RIOT/RIOT/examples/ng_networking/udp.c:88:48: warning: format specifies type 'int' but the argument has type
      'size_t' (aka 'unsigned long') [-Wformat]
    printf("Success: send %i byte to %s:%u\n", payload->size, addr_str, tmp);
                          ~~                   ^~~~~~~~~~~~~
                          %zu
1 warning generated.
"/Applications/Xcode.app/Contents/Developer/usr/bin/make" -C /Users/emmanuelB/Documents/work/CODING/RIOT/RIOT/boards/native
"/Applications/Xcode.app/Contents/Developer/usr/bin/make" -C /Users/emmanuelB/Documents/work/CODING/RIOT/RIOT/boards/native/drivers
"/Applications/Xcode.app/Contents/Developer/usr/bin/make" -C /Users/emmanuelB/Documents/work/CODING/RIOT/RIOT/core
"/Applications/Xcode.app/Contents/Developer/usr/bin/make" -C /Users/emmanuelB/Documents/work/CODING/RIOT/RIOT/cpu/native
/Users/emmanuelB/Documents/work/CODING/RIOT/RIOT/cpu/native/hwtimer_cpu.c:278:5: warning: implicit declaration of function 'host_get_clock_service' is
      invalid in C99 [-Wimplicit-function-declaration]
    host_get_clock_service(mach_host_self(), SYSTEM_CLOCK, &cclock);
    ^
/Users/emmanuelB/Documents/work/CODING/RIOT/RIOT/cpu/native/hwtimer_cpu.c:278:28: warning: implicit declaration of function 'mach_host_self' is
      invalid in C99 [-Wimplicit-function-declaration]
    host_get_clock_service(mach_host_self(), SYSTEM_CLOCK, &cclock);
                           ^
/Users/emmanuelB/Documents/work/CODING/RIOT/RIOT/cpu/native/hwtimer_cpu.c:280:5: warning: implicit declaration of function 'mach_port_deallocate' is
      invalid in C99 [-Wimplicit-function-declaration]
    mach_port_deallocate(mach_task_self(), cclock);
    ^
/Users/emmanuelB/Documents/work/CODING/RIOT/RIOT/cpu/native/hwtimer_cpu.c:280:26: warning: implicit declaration of function 'mach_task_self' is
      invalid in C99 [-Wimplicit-function-declaration]
    mach_port_deallocate(mach_task_self(), cclock);
                         ^
4 warnings generated.
/Users/emmanuelB/Documents/work/CODING/RIOT/RIOT/cpu/native/irq_cpu.c:324:5: warning: 'makecontext' is deprecated: first deprecated in OS X 10.6
      [-Wdeprecated-declarations]
    makecontext(&native_isr_context, native_irq_handler, 0);
    ^
/usr/include/ucontext.h:38:6: note: 'makecontext' has been explicitly marked deprecated here
void makecontext(ucontext_t *, void (*)(), int, ...) __OSX_AVAILABLE_BUT_DEPRECATED(__MAC_10_5, __MAC_10_6, __IPHONE_2_0, __IPHONE_2_0);
     ^
/Users/emmanuelB/Documents/work/CODING/RIOT/RIOT/cpu/native/irq_cpu.c:486:9: warning: 'getcontext' is deprecated: first deprecated in OS X 10.6
      [-Wdeprecated-declarations]
    if (getcontext(&native_isr_context) == -1) {
        ^
/usr/include/ucontext.h:37:6: note: 'getcontext' has been explicitly marked deprecated here
int  getcontext(ucontext_t *) __OSX_AVAILABLE_BUT_DEPRECATED(__MAC_10_5, __MAC_10_6, __IPHONE_2_0, __IPHONE_2_0);
     ^
/Users/emmanuelB/Documents/work/CODING/RIOT/RIOT/cpu/native/irq_cpu.c:504:5: warning: 'makecontext' is deprecated: first deprecated in OS X 10.6
      [-Wdeprecated-declarations]
    makecontext(&native_isr_context, native_irq_handler, 0);
    ^
/usr/include/ucontext.h:38:6: note: 'makecontext' has been explicitly marked deprecated here
void makecontext(ucontext_t *, void (*)(), int, ...) __OSX_AVAILABLE_BUT_DEPRECATED(__MAC_10_5, __MAC_10_6, __IPHONE_2_0, __IPHONE_2_0);
     ^
3 warnings generated.
/Users/emmanuelB/Documents/work/CODING/RIOT/RIOT/cpu/native/native_cpu.c:118:9: warning: 'getcontext' is deprecated: first deprecated in OS X 10.6
      [-Wdeprecated-declarations]
    if (getcontext(p) == -1) {
        ^
/usr/include/ucontext.h:37:6: note: 'getcontext' has been explicitly marked deprecated here
int  getcontext(ucontext_t *) __OSX_AVAILABLE_BUT_DEPRECATED(__MAC_10_5, __MAC_10_6, __IPHONE_2_0, __IPHONE_2_0);
     ^
/Users/emmanuelB/Documents/work/CODING/RIOT/RIOT/cpu/native/native_cpu.c:131:5: warning: 'makecontext' is deprecated: first deprecated in OS X 10.6
      [-Wdeprecated-declarations]
    makecontext(p, (void (*)(void)) task_func, 1, arg);
    ^
/usr/include/ucontext.h:38:6: note: 'makecontext' has been explicitly marked deprecated here
void makecontext(ucontext_t *, void (*)(), int, ...) __OSX_AVAILABLE_BUT_DEPRECATED(__MAC_10_5, __MAC_10_6, __IPHONE_2_0, __IPHONE_2_0);
     ^
[... 114 lines cut ...]
    ^
/Users/emmanuelB/Documents/work/CODING/RIOT/RIOT/cpu/native/ng_net/dev_eth_tap.c:273:26: error: unknown type name 'ng_tapnet_t'
static void _sigio_child(ng_tapnet_t *dev)
                         ^
/Users/emmanuelB/Documents/work/CODING/RIOT/RIOT/cpu/native/ng_net/dev_eth_tap.c:276:10: error: use of undeclared identifier '_sigio_child_pid'
    if ((_sigio_child_pid = real_fork()) == -1) {
         ^
/Users/emmanuelB/Documents/work/CODING/RIOT/RIOT/cpu/native/ng_net/dev_eth_tap.c:279:9: error: use of undeclared identifier '_sigio_child_pid'; did
      you mean '_sigio_child'?
    if (_sigio_child_pid > 0) {
        ^~~~~~~~~~~~~~~~
        _sigio_child
[... cut ...]
      ('int (*)()' and 'int') is an extension [-Wpedantic]
    if (_sigio_child_pid > 0) {
        ~~~~~~~~~~~~~~~~ ^ ~
2 warnings and 14 errors generated.
make[3]: *** [/Users/emmanuelB/Documents/work/CODING/RIOT/RIOT/examples/ng_networking/bin/native/ng_nativenet/dev_eth_tap.o] Error 1
make[2]: *** [ALL--/Users/emmanuelB/Documents/work/CODING/RIOT/RIOT/cpu/native/ng_net] Error 2
make[1]: *** [ALL--/Users/emmanuelB/Documents/work/CODING/RIOT/RIOT/cpu/native] Error 2
make: *** [all] Error 2
```

emmanuelsearch, 2015-07-02:
It also compiles for ng_networking example, but I also get quite some warnings (see below)

```
$ make -B clean all
Configured with: --prefix=/Applications/Xcode.app/Contents/Developer/usr --with-gxx-include-dir=/usr/include/c++/4.2.1
Building application "ng_networking" for "native" with MCU "native".

/Users/emmanuelB/Documents/work/CODING/RIOT/RIOT/examples/ng_networking/udp.c:88:48: warning: 
      format specifies type 'int' but the argument has type 'size_t'
      (aka 'unsigned long') [-Wformat]
    printf("Success: send %i byte to %s:%u\n", payload->size, addr_str, tmp);
                          ~~                   ^~~~~~~~~~~~~
                          %zu
1 warning generated.
...
/Users/emmanuelB/Documents/work/CODING/RIOT/RIOT/cpu/native
/Users/emmanuelB/Documents/work/CODING/RIOT/RIOT/cpu/native/hwtimer_cpu.c:280:5: warning: 
      implicit declaration of function 'host_get_clock_service' is invalid in
      C99 [-Wimplicit-function-declaration]
    host_get_clock_service(mach_host_self(), SYSTEM_CLOCK, &cclock);
    ^
1 warning generated.
/Users/emmanuelB/Documents/work/CODING/RIOT/RIOT/cpu/native/irq_cpu.c:324:5: warning: 
      'makecontext' is deprecated: first deprecated in OS X 10.6
      [-Wdeprecated-declarations]
    makecontext(&native_isr_context, native_irq_handler, 0);
    ^
/usr/include/ucontext.h:38:6: note: 'makecontext' has been explicitly marked
      deprecated here
void makecontext(ucontext_t *, void (*)(), int, ...) __OSX_AVAILABLE_BUT...
     ^
/Users/emmanuelB/Documents/work/CODING/RIOT/RIOT/cpu/native/irq_cpu.c:486:9: warning: 
      'getcontext' is deprecated: first deprecated in OS X 10.6
      [-Wdeprecated-declarations]
    if (getcontext(&native_isr_context) == -1) {
        ^
/usr/include/ucontext.h:37:6: note: 'getcontext' has been explicitly marked
      deprecated here
int  getcontext(ucontext_t *) __OSX_AVAILABLE_BUT_DEPRECATED(__MAC_10_5...
     ^
/Users/emmanuelB/Documents/work/CODING/RIOT/RIOT/cpu/native/irq_cpu.c:504:5: warning: 
      'makecontext' is deprecated: first deprecated in OS X 10.6
      [-Wdeprecated-declarations]
    makecontext(&native_isr_context, native_irq_handler, 0);
    ^
/usr/include/ucontext.h:38:6: note: 'makecontext' has been explicitly marked
      deprecated here
void makecontext(ucontext_t *, void (*)(), int, ...) __OSX_AVAILABLE_BUT...
     ^
3 warnings generated.
/Users/emmanuelB/Documents/work/CODING/RIOT/RIOT/cpu/native/native_cpu.c:118:9: warning: 
      'getcontext' is deprecated: first deprecated in OS X 10.6
      [-Wdeprecated-declarations]
    if (getcontext(p) == -1) {
        ^
/usr/include/ucontext.h:37:6: note: 'getcontext' has been explicitly marked
      deprecated here
int  getcontext(ucontext_t *) __OSX_AVAILABLE_BUT_DEPRECATED(__MAC_10_5...
     ^
/Users/emmanuelB/Documents/work/CODING/RIOT/RIOT/cpu/native/native_cpu.c:131:5: warning: 
      'makecontext' is deprecated: first deprecated in OS X 10.6
      [-Wdeprecated-declarations]
    makecontext(p, (void (*)(void)) task_func, 1, arg);
    ^
/usr/include/ucontext.h:38:6: note: 'makecontext' has been explicitly marked
      deprecated here
void makecontext(ucontext_t *, void (*)(), int, ...) __OSX_AVAILABLE_BUT...
     ^
/Users/emmanuelB/Documents/work/CODING/RIOT/RIOT/cpu/native/native_cpu.c:153:9: warning: 
      'setcontext' is deprecated: first deprecated in OS X 10.6
[... 180 lines cut ...]
  __builtin___sprintf_chk (str, 0, __darwin_obsz(str), __VA_ARGS__)
                                                       ^
/Users/emmanuelB/Documents/work/CODING/RIOT/RIOT/sys/od/od.c:169:41: warning: 
      format specifies type 'int' but the argument has type 'unsigned long'
      [-Wformat]
            sprintf(format, " %%0%dhx", sizeof(short) * _HEX_BYTE_LENGTH);
            ~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
                                 %lu
/usr/include/secure/_stdio.h:47:56: note: expanded from macro 'sprintf'
  __builtin___sprintf_chk (str, 0, __darwin_obsz(str), __VA_ARGS__)
                                                       ^
/Users/emmanuelB/Documents/work/CODING/RIOT/RIOT/sys/od/od.c:173:41: warning: 
      format specifies type 'int' but the argument has type 'unsigned long'
      [-Wformat]
            sprintf(format, " %%0%dlx", sizeof(long) * _HEX_BYTE_LENGTH);
            ~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
                                 %lu
/usr/include/secure/_stdio.h:47:56: note: expanded from macro 'sprintf'
  __builtin___sprintf_chk (str, 0, __darwin_obsz(str), __VA_ARGS__)
                                                       ^
8 warnings generated.
...
__TEXT	__DATA	__OBJC	others	dec	hex
159744	1888256	0	36864	2084864	1fd000
```

LudwigOrtmann, 2015-07-11:
@thomaseichinger not off the top of my hat, no time for deeper soul search atm

OlegHahm, 2015-07-13:
> @thomaseichinger not off the top of my hat

:tophat: ?

OlegHahm, 2015-09-03:
@emmanuelsearch, @thomaseichinger, is this still the case?

thomaseichinger, 2015-09-03:
Fixed by #3393

closed 2015-09-03 by thomaseichinger